minosworld/minos #99: Continuous control

ducha-aiki, 2018-09-21:
Hi,

I am working on PR, enabling continuous control in gym env.

So far I have added "Box" state for actions here 
https://github.com/minosworld/minos/blob/master/gym/gym_minos/envs/indoor_env.py#L28

change here 
https://github.com/minosworld/minos/blob/master/minos/lib/RoomSimulator.py#L31

and send actual strength instead of hardcoded here:
https://github.com/minosworld/minos/blob/1b1d560bbfc613d12ef947c43c2864ca14c9b586/minos/lib/RoomSimulator.py#L185

But the problem is that I observe no difference, regardless of strength I sent to js part. Could you please tell me, what is needed to be fixed on javascript side?

angelxuanchang, 2018-10-11:
To use the strength parameters, you will need to update the [FirstPersonAgent.js](https://github.com/smartscenes/sstk/blob/master/client/js/lib/sim/FirstPersonAgent.js) to use the strength field (see lookUp/lookDown for example).

ducha-aiki, 2018-10-11:
@angelxuanchang thanks, I`ll give it a try!

msavva, 2018-10-11:
@ducha-aiki One note about making changes to the javascript side: the sstk code will need to be re-built after making changes.  Probably the most convenient way to develop with local changes in sstk is to use a local checkout of sstk in https://github.com/minosworld/minos/blob/master/minos/server/package.json#L19 
You can replace the quoted sstk package version with something like "~/path/to/sstk/".  Then you can make changes in that local checkout of sstk and re-build using the `build.sh` script or by running `NODE_ENV=dev npm run build` in `~/path/to/sstk/client` to have continuous rebuilding.