minosworld/minos #100: Look up and look down are not handled in agent_state

ducha-aiki, 2018-09-22:
Hi,

I think, it is worth mentioning in API that agent and camera are two very different entities. One of the limitations that that when camera is looked up or down, agent orientation in info['agent_state']['orientation'] is not changed.
Why does anyone ever need to look down? Sometimes, because of limited vertical field of view, nearby obstacle is not visible and "go backwards"  to see it is not possible due to tight space constraints. One way of see (and possible map) the obstacle is to look down and map it, e.g. via depth map. But agent orientation is wrong, so it cannot be done directly. 

Moreover, even if one fix the orientation, then camera height is changing when look up and down as well. 
So, it is probably not worth to fix it somehow (unless report camera position instead of agent), but worth to somehow mention this in docs

angelxuanchang, 2018-10-11:
Yes, the "head"/"camera" orientation is different of the overall agent orientation.  We will consider improving this for the next release.  And indeed we should clarify the docs and also add camera orientation to output.

mfadhilgtg, 2019-03-26:
Hi, have you released the camera orientation output in the release 0.7x? Or could you give hint where can we access the camera orientation. Thank you