minosworld/minos #101: How to run my own 3D model in minos ?

Usama323, 2018-09-24:
I have my own 3D model which contain a ".obj" , ".mtl" and ".jpg" files , can I run my own model in minos environment ?

Usama323, 2018-09-24:
if we could include model of a house/room that would be great , I want to add it in matterport data! if I can do it kindly explain its procedure.

angelxuanchang, 2018-10-10:
I'm not sure if it makes sense to include a model of a house or room in the matterport data since the matterport data is reconstruction of an existing house.

Usama323, 2018-10-11:
can we add model of house/room in suncg data ?

angelxuanchang, 2018-10-11:
Not sure if it makes that much sense either since SUNCG has it's own definition of rooms.  Please see https://github.com/shurans/SUNCGtoolbox for how the SUNCG data is organized and the format of the house.json file. You can modify / introduce a new house.json that has your house/room.  

But is do you want to add your house/room to an existing environment?  Or do you just want to treat your model as another environment from a different dataset?  Those are different things.