minosworld/minos #102: Can I change brightness for one room?

Usama323, 2018-09-24:
Is it possible to change the brightness for one room (increase/decrease) it ?

angelxuanchang, 2018-09-24:
Is this something you want to do for SUNCG rooms?

Usama323, 2018-09-24:
In which environment I can do it ? (matterport or SUNCG ). And what is method to do it ?

Usama323, 2018-09-24:
can we implement this in materport data set  (house)

ZhuFengdaaa, 2018-10-10:
Minos has nothing to do with it, but [SUNCGtoolbox](https://github.com/shurans/SUNCGtoolbox) provide some API about it. Referring to [this metadata file](https://github.com/shurans/SUNCGtoolbox/blob/master/metadata/suncgModelLights.json), you can configure the power(I think this defines brightness), color, type and position of the light. 

By the way, it would be better do some investigation before you ask for help. Authors are busy as well :-)

angelxuanchang, 2018-10-10:
Thanks @ZhuFengdaaa.  Minos currently does not support varying the lighting.  It is something we plan to support in the future.