minosworld/minos #104: How to get 45K House3D houses

ZhuFengdaaa, 2018-10-01:
Both the [minos](https://arxiv.org/pdf/1712.03931.pdf) and [house3d](https://arxiv.org/pdf/1801.02209.pdf) claim that there is 45K houses in suncg dataset, which provide rich training data. 

However, [data file in this project](https://github.com/minosworld/minos/blob/master/minos/data/scenes.multiroom.csv) only provides 365 different houses. Does anybody know how to get 45K different scenes ?

angelxuanchang, 2018-10-03:
Once you download the SUNCG datasets, you will have a house directory with <id>/house.json.  Each of those house file correspond to a different scene (there should be 45622 in total - not necessarily all are appropriate for the navigation task).   You can get the ids by listing the house directory.

ZhuFengdaaa, 2018-10-03:
It means that I can use [minos/tools/generate_episodes.py](https://github.com/minosworld/minos/blob/master/minos/tools/generate_episodes.py) to prepare dataset myself, right?

May I ask why there so many houses are not appropriate for navigation task that both minos and house3D papers just select a few houses manually for experiment? Too complex, too large or something else?

ZhuFengdaaa, 2018-10-03:
Thank you for your detailed response. I hope that we can find way to utilise 3D resource as much as possible and get a larger and more diverse dataset. For example, sample episode from a subset of rooms connected with each other can overcome 1) and 3).

angelxuanchang, 2018-10-10:
Yes, it is good to leverage more of the dataset.  We welcome contributions!  Please let us know if you have any other questions or issues with using the larger dataset.

closed 2018-10-10 by angelxuanchang

liucsg, 2020-03-06:
@angelxuanchang Hi，can i use personal email for requesting access to download datasets，including matterport3d?