minosworld/minos #108: Errors after running npm install

GreyCoder25, 2018-10-11:
Hello, I wanted to try minos but after running npm install in the server directory got the next errors:
failed: true,
  signal: null,
  cmd: 'npm install',
  timedOut: false,
  exitCode: 1 }
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! sstk@0.5.3 bootstrap: `lerna bootstrap --hoist`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the sstk@0.5.3 bootstrap script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/serhii/.npm/_logs/2018-10-11T21_10_27_100Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! sstk@0.5.3 postinstall: `npm run bootstrap`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the sstk@0.5.3 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/serhii/.npm/_logs/2018-10-11T21_10_28_990Z-debug.log

Full output was next:

> sstk@0.5.3 preinstall /home/serhii/work/minos/minos/server/node_modules/sstk
> ./getdata.sh

Cloning into './metadata'...
remote: Enumerating objects: 29, done.
remote: Counting objects: 100% (29/29), done.
remote: Compressing objects: 100% (18/18), done.
remote: Total 94 (delta 23), reused 11 (delta 11), pack-reused 65
Unpacking objects: 100% (94/94), done.
Checking connectivity... done.

> node-expat@2.3.16 install /home/serhii/work/minos/minos/server/node_modules/node-expat
> node-gyp rebuild

make: Entering directory '/home/serhii/work/minos/minos/server/node_modules/node-expat/build'
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmlparse.o
../deps/libexpat/lib/xmlparse.c: In function ‘gather_time_entropy’:
../deps/libexpat/lib/xmlparse.c:780:7: warning: variable ‘gettimeofday_res’ set but not used [-Wunused-but-set-variable]
   int gettimeofday_res;
       ^
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmltok.o
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmlrole.o
  AR(target) Release/obj.target/deps/libexpat/libexpat.a
  COPY Release/libexpat.a
  CXX(target) Release/obj.target/node_expat/node-expat.o
In file included from ../../nan/nan_new.h:189:0,
                 from ../../nan/nan.h:203,
                 from ../node-expat.cc:1:
../../nan/nan_implementation_12_inl.h: In static member function ‘static Nan::imp::FactoryBase<v8::StringObject>::return_t Nan::imp::Factory<v8::StringObject>::New(v8::Local<v8::String>)’:
../../nan/nan_implementation_12_inl.h:337:37: warning: ‘static v8::Local<v8::Value> v8::StringObject::New(v8::Local<v8::String>)’ is deprecated: Use Isolate* version [-Wdeprecated-declarations]
   return v8::StringObject::New(value).As<v8::StringObject>();
                                     ^
In file included from /home/serhii/.node-gyp/10.12.0/include/node/v8.h:26:0,
                 from /home/serhii/.node-gyp/10.12.0/include/node/node.h:63,
                 from ../../nan/nan.h:51,
                 from ../node-expat.cc:1:
/home/serhii/.node-gyp/10.12.0/include/node/v8.h:5053:37: note: declared here
                 static Local<Value> New(Local<String> value));
                                     ^
/home/serhii/.node-gyp/10.12.0/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
[... 704 lines cut ...]
  failed: true,
  signal: null,
  cmd: 'npm install',
  timedOut: false,
  exitCode: 1 }
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! sstk@0.5.3 bootstrap: `lerna bootstrap --hoist`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the sstk@0.5.3 bootstrap script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/serhii/.npm/_logs/2018-10-11T21_10_27_100Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! sstk@0.5.3 postinstall: `npm run bootstrap`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the sstk@0.5.3 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/serhii/.npm/_logs/2018-10-11T21_10_28_990Z-debug.log

msavva, 2018-10-11:
One error that might be the root cause is failure to download https://dl.bintray.com/lovell/sharp/libvips-8.5.5-linux-x64.tar.gz (due to certificate issues).  Can you confirm if you have installed the prerequisite dependencies (see step 3 in https://github.com/minosworld/minos#installing) before invoking `npm install`?

GreyCoder25, 2018-10-12:
Yes, I've installed them.

GreyCoder25, 2018-10-12:
Yes, I had installed all this stuff before invoking npm install.

angelxuanchang, 2018-10-12:
Please check that you have libvips installed.  The installation is failing because it cannot find libvips and is trying to download: https://dl.bintray.com/lovell/sharp/libvips-8.5.5-linux-x64.tar.gz 

You can also check that you can download https://dl.bintray.com/lovell/sharp/libvips-8.5.5-linux-x64.tar.gz  from a browser.  And you can try `NODE_TLS_REJECT_UNAUTHORIZED=0 npm install` to allow download of https://dl.bintray.com/lovell/sharp/libvips-8.5.5-linux-x64.tar.gz  (see [nodejs/node-gyp#695](https://github.com/nodejs/node-gyp/issues/695))

GreyCoder25, 2018-10-12:
Yes, thank you, possibly it's not the best way but it solved the problem.

closed 2018-10-12 by GreyCoder25

GreyCoder25, 2018-10-12:
Also it's worth to add that after running "NODE_TLS_REJECT_UNAUTHORIZED=0 npm install" end of the output is next: 

added 286 packages from 118 contributors and audited 1741 packages in 264.012s
found 31 vulnerabilities (17 low, 8 moderate, 5 high, 1 critical)
  run `npm audit fix` to fix them, or `npm audit` for details

And after invoking "npm audit fix" minos pygame client gives a connection problem:

2018-10-12 19:03:20,494 INFO sim00:sim server has exited with rv 1
2018-10-12 19:03:20,506 WARNING localhost:56954/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=56954): Max retries exceeded with url: /socket.io/?transport=polling&t=1539360200496-0&EIO=3 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f9fa9c57b38>: Failed to establish a new connection: [Errno 111] Connection refused',)).

But ALL WORKS if NOT run "npm audit fix".

angelxuanchang, 2018-10-12:
Glad that it worked.  Yes, please don't run the `npm audit fix` .  Some of the packages that are used are pretty old and we are working on minos 0.7.x to update to new packages and fix this issue.

TopCoder2K, 2021-11-24:
@GreyCoder25, could you tell me the steps you've done to fix the problem? I'm having the same error, but just running `NODE_TLS_REJECT_UNAUTHORIZED=0 npm install` haven't fixed it :(

TopCoder2K, 2021-11-24:
And also the links kindly provided by @angelxuanchang don't work, I get "forbidden"