minosworld/minos #109: IndexError: index out of range

wqynew, 2018-10-13:
**My problem is**
Starting simulator...
2018-10-13 15:18:59,573 INFO sim00:Starting sim server at /home/exbot/minos/minos/server/server.js with port 39142
Traceback (most recent call last):
File "/home/exbot/minos/minos/tools/pygame_client.py", line 438, in main
ep_info = sim.start()
File "/home/exbot/minos/minos/lib/Simulator.py", line 321, in start
started = self.start_child_servers()
File "/home/exbot/minos/minos/lib/Simulator.py", line 255, in start_child_servers
self._sio = SocketIO(self.params.host, self.params.port)
File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/init.py", line 345, in init
resource, hurry_interval_in_seconds, **kw)
File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/init.py", line 58, in init
self._transport
File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/init.py", line 66, in _transport
self._engineIO_session = self._get_engineIO_session()
File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/init.py", line 80, in _get_engineIO_session
transport.recv_packet())
File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/transports.py", line 92, in recv_packet
for engineIO_packet in decode_engineIO_content(response.content):
File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/parsers.py", line 95, in decode_engineIO_content
content, content_index)
File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/parsers.py", line 202, in _read_packet_length
while get_byte(content, content_index) not in [0, 1]:
File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/symmetries.py", line 28, in get_byte
return six.indexbytes(x, index)
IndexError: index out of range
Error running simulator. Aborting.
2018-10-13 15:19:02,499 INFO sim00:Stopping the simulator
2018-10-13 15:19:02,499 INFO Counter()
2018-10-13 15:19:02,499 INFO sim00:Stopping child servers
2018-10-13 15:19:02,499 INFO sim00:Killing sim pid 8022
2018-10-13 15:19:02,506 INFO sim00:Stopped child servers
2018-10-13 15:19:02,506 INFO sim00:Simulator killed.
**the simserver.log:**
2018-10-13 15:19:00,195 INFO b'Configuration { fileCache: { size: 50 },'
2018-10-13 15:19:00,195 INFO b' imageCache: { size: 200 },'
2018-10-13 15:19:00,195 INFO b' imageQueue: { concurrency: 4 },'
2018-10-13 15:19:00,195 INFO b" base_url: '/home/exbot/work/',"
2018-10-13 15:19:00,195 INFO b' assets_url:'
2018-10-13 15:19:00,195 INFO b" 'file:///home/exbot/minos/minos/server/node_modules/sstk/ssc/../server/static' }"
2018-10-13 15:19:00,454 INFO b'Waiting for client connection on port 39142'
2018-10-13 15:19:00,476 ERROR b'Cannot register unknown asset 2d3ds'
2018-10-13 15:19:00,477 INFO b'register ../server/static/data/matterport/matterport3d.json'
2018-10-13 15:19:00,480 INFO b'Registered asset group: mp3d'
2018-10-13 15:19:00,485 INFO b'Got 90 assets'
2018-10-13 15:19:00,558 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2018-10-13 15:19:00,559 INFO b'Replace asset group p5d'
2018-10-13 15:19:00,560 INFO b'Registered asset group: p5d'
2018-10-13 15:19:00,681 INFO b'Got 2623 assets'
2018-10-13 15:19:01,681 INFO b'Loaded lightSpecs for 202 models.'
2018-10-13 15:19:01,681 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2018-10-13 15:19:01,683 INFO b'Replace asset group p5dTexture'
2018-10-13 15:19:01,683 INFO b'Registered asset group: p5dTexture'
2018-10-13 15:19:02,151 INFO b'Got 67072 assets'
2018-10-13 15:19:02,499 INFO Waiting for simserver stdout to finish
2018-10-13 15:19:02,505 INFO Finished simserver stdout
2018-10-13 15:19:02,506 INFO Waiting for simserver stderr to finish
2018-10-13 15:19:02,506 INFO Finished simserver stderr
**the simulator.log**
2018-10-13 15:18:59,553 INFO {'sim_id': 'sim00', 'machine': '123', 'sim_git_hash': 'c66acdb', 'stk_git_hash': 'c66acdb'}
2018-10-13 15:18:59,573 INFO sim00:Starting sim server at /home/exbot/minos/minos/server/server.js with port 39142
2018-10-13 15:19:02,499 INFO sim00:Stopping the simulator
2018-10-13 15:19:02,499 INFO Counter()
2018-10-13 15:19:02,499 INFO sim00:Stopping child servers
2018-10-13 15:19:02,499 INFO sim00:Killing sim pid 8022
2018-10-13 15:19:02,506 INFO sim00:Stopped child servers
2018-10-13 15:19:02,506 INFO sim00:Simulator killed.

Could you please give me some advice?

wqynew, 2018-10-14:
@ZhuFengdaaa Thanks for your reply. However, (1) my problem is different from #39, which is caused by the wrong SUNCG dataset. I still don't know what's wrong with the configuration in my computer (a Virtual Machine, ubuntu 14.04). Can you give me some advice? (2) I don't have the Matterport3D dataset currently, though I have written an email to Matterport3D institute three days ago. Do you think it's a normal or should I write again? 
Any response will be highly appreciated. Thanks again.

msavva, 2018-10-14:
@wqynew this issue is due to an version mismatch for the socketIO client package which is a dependency.  Ways to resolve this issue are described in https://github.com/minosworld/minos/blob/master/FAQ.md#i-get-an-error-from-the-socketio_client-package .  However, we have recently updated the code to prevent this from happening.  Please try to use the code on the `v0.7.x` branch which includes an update that forces the correct version of the dependency to be downloaded.  Let us know if you encounter any problems.

wqynew, 2018-11-21:
Sorry for replying so late. It can work now. Thanks very much.@msavva