minosworld/minos #110: Wired mp3d images full of fragments receive from gym

ZhuFengdaaa, 2018-10-13:
I sample an episode from scene  with `gym/gym_minos/envs/indoor_env.py`, which video is shown [here](https://drive.google.com/file/d/13nDs0wDobmswoFq0xGQkjqgQ6xgOyZDF/view?usp=sharing). 

However when using `vglrun python3 -m minos.tools.pygame_client --env_config roomgoal_mp3d_s --scene_ids sT4fr6TAbpF` to show the scene, the images are clear without any fragment.

msavva, 2018-10-14:
@ZhuFengdaaa thanks for reporting this.  Can you also provide a link to the `simulator.log` and `simserver.log` files that are created when you record a video with these artifacts?

ZhuFengdaaa, 2018-10-15:
[simserver.log](https://paste.ubuntu.com/p/szWFPdSmF5/) and [simulator.log](https://paste.ubuntu.com/p/XxG7Zfdryp/).

ducha-aiki, 2018-11-27:
@msavva @ZhuFengdaaa I confirm this, get the same issue and actually never seen "clean" version, so thought that it is texture problem of the specific map.

ZhuFengdaaa, 2018-12-14:
Still waiting for solution. If this issue is not fixed, all experiments on MINOS mp3d would be unconvincing.

angelxuanchang, 2018-12-15:
So far, we have not been able to reproduce this issue.  Can you provide more information about when you see this?  Is this only after running the agent for a long time?  Does it happen at specific viewpoints?

ZhuFengdaaa, 2019-01-11:
Sorry for the delayed response. This issue is happen very often. I can reproduce it on most scene EXCEPT `17DRP5sb8fy`. It don't need to run agent for a long time and also happened on many normal viewpoints. As I describe above, I use the command 
```
vglrun python3 -m minos.tools.pygame_client --env_config roomgoal_mp3d_s --scene_ids sT4fr6TAbpF
```
to generate video shown lots of fragments.

msavva, 2019-01-21:
Hi @ZhuFengdaaa , can you try to use the EGL version of headless-gl as described here: https://github.com/smartscenes/sstk/wiki/Headless-Rendering#using-forked-headless-gl-with-egl-support instead of using VirtualGL?  Some of the rendering issues might stem from the fact that running through VirtualGL makes the OpenGL context acquisition more complex.  Let us know if you observe this issue still when using EGL, or when running on a local machine (that does not require using VirtualGL).

ducha-aiki, 2019-01-22:
@msavva I have used headless-gl-nvidia you provided and never used Virtual GL. Nevertheless, I have experienced the same issues.

ZhuFengdaaa, 2019-01-30:
@msavva 
I run without `virtualGL` 
```python3 -m minos.tools.pygame_client --env_config roomgoal_mp3d_s --scene_ids sT4fr6TAbpF``` , writing images to disk by "cv2.imwrite", and experienced the same issue.

ZhuFengdaaa, 2019-03-05:
Still waiting for solution..Same experience when I use  headless-gl-nvidia. @ducha-aiki Have you solved this problem ?

ducha-aiki, 2019-03-17:
@ZhuFengdaaa no