minosworld/minos #121: Problem while running MINOS

romi-seecs, 2018-11-14:
Hi! Following error occurs while running MINOS. I tried reinstalling Python3 but it had no effect, I've installed Anaconda as well.

```
romi@romi-Z370-AORUS-Gaming-3:~/minosmi/minos$ python3 -m minos.tools.pygame_client --empty_room
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
2018-11-14 19:44:31,043 INFO {'sim_id': 'sim00', 'machine': 'romi-Z370-AORUS-Gaming-3', 'sim_git_hash': '478f622', 'stk_git_hash': '478f622'}
Starting simulator...
2018-11-14 19:44:31,051 INFO sim00:Starting sim server at /home/romi/minosmi/minos/minos/server/server.js with port 51849
2018-11-14 19:44:32,055 INFO sim00:sim server has exited with rv 1
2018-11-14 19:44:32,062 WARNING localhost:51849/socket.io [engine.io waiting for connection] HTTPConnectionPool(host='localhost', port=51849): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1542206672057-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa3ce361668>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-11-14 19:44:33,066 WARNING localhost:51849/socket.io [engine.io waiting for connection] HTTPConnectionPool(host='localhost', port=51849): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1542206673063-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa3ce3614a8>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-11-14 19:44:34,070 WARNING localhost:51849/socket.io [engine.io waiting for connection] HTTPConnectionPool(host='localhost', port=51849): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1542206674067-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa3ce3615f8>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-11-14 19:44:35,074 WARNING localhost:51849/socket.io [engine.io waiting for connection] HTTPConnectionPool(host='localhost', port=51849): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1542206675071-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa3ce361e10>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-11-14 19:44:36,078 WARNING localhost:51849/socket.io [engine.io waiting for connection] HTTPConnectionPool(host='localhost', port=51849): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1542206676076-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa3ce361cc0>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-11-14 19:44:37,082 WARNING localhost:51849/socket.io [engine.io waiting for connection] HTTPConnectionPool(host='localhost', port=51849): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1542206677080-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa3ce361630>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-11-14 19:44:38,086 WARNING localhost:51849/socket.io [engine.io waiting for connection] HTTPConnectionPool(host='localhost', port=51849): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1542206678084-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa3ce361668>: Failed to establish a new connection: [Errno 111] Connection refused',))
^CTraceback (most recent call last):
  File "/home/romi/anaconda3/lib/python3.6/site-packages/socketIO_client/logs.py", line 29, in _yield_warning_screen
    yield elapsed_time
Exception: [engine.io waiting for connection] HTTPConnectionPool(host='localhost', port=51849): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1542206678084-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fa3ce361668>: Failed to establish a new connection: [Errno 111] Connection refused',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/romi/minosmi/minos/minos/tools/pygame_client.py", line 438, in main
    ep_info = sim.start()
  File "/home/romi/minosmi/minos/minos/lib/Simulator.py", line 321, in start
    started = self.start_child_servers()
  File "/home/romi/minosmi/minos/minos/lib/Simulator.py", line 255, in start_child_servers
    self._sio = SocketIO(self.params.host, self.params.port)
  File "/home/romi/anaconda3/lib/python3.6/site-packages/socketIO_client/__init__.py", line 353, in __init__
    resource, hurry_interval_in_seconds, **kw)
  File "/home/romi/anaconda3/lib/python3.6/site-packages/socketIO_client/__init__.py", line 54, in __init__
    self._transport
  File "/home/romi/anaconda3/lib/python3.6/site-packages/socketIO_client/__init__.py", line 62, in _transport
    self._engineIO_session = self._get_engineIO_session()
  File "/home/romi/anaconda3/lib/python3.6/site-packages/socketIO_client/__init__.py", line 83, in _get_engineIO_session
    warning_screen.throw(warning)
  File "/home/romi/anaconda3/lib/python3.6/site-packages/socketIO_client/logs.py", line 35, in _yield_warning_screen
    time.sleep(1)
KeyboardInterrupt
Error running simulator. Aborting.
2018-11-14 19:44:38,210 INFO sim00:Stopping the simulator
2018-11-14 19:44:38,210 INFO Counter()
2018-11-14 19:44:38,210 INFO sim00:Stopping child servers
2018-11-14 19:44:38,211 INFO sim00:Stopped child servers
2018-11-14 19:44:38,211 INFO sim00:Simulator killed.

```

jsdd25, 2018-11-27:
@romi-seecs Hi, I have met the same problem with you. Have you fixed the problem? Thank you very much!

ZhuFengdaaa, 2018-11-28:
Please attach the ` log/simserver.log` to see whats the problem.

jsdd25, 2018-11-28:
@ZhuFengdaaa Hi, That is my simserver.log. Please give me some help. Thank you very much!
### **simserver.log:**
2018-11-28 11:28:21,215 ERROR b'/home/jin/minos/minos/server/node_modules/sstk/ssc/stk-ssc.js:25'
2018-11-28 11:28:21,215 ERROR b'var doc = jsdom.jsdom(\'<!doctype html><html><head></head><body><div id="canvas"></div></body></html>\', {'
2018-11-28 11:28:21,215 ERROR b'                ^'
2018-11-28 11:28:21,215 ERROR b''
2018-11-28 11:28:21,215 ERROR b'TypeError: jsdom.jsdom is not a function'
2018-11-28 11:28:21,215 ERROR b'    at Object.<anonymous> (/home/jin/minos/minos/server/node_modules/sstk/ssc/stk-ssc.js:25:17)'
2018-11-28 11:28:21,215 ERROR b'    at Module._compile (internal/modules/cjs/loader.js:688:30)'
2018-11-28 11:28:21,216 ERROR b'    at Object.Module._extensions..js (internal/modules/cjs/loader.js:699:10)'
2018-11-28 11:28:21,216 ERROR b'    at Module.load (internal/modules/cjs/loader.js:598:32)'
2018-11-28 11:28:21,216 ERROR b'    at tryModuleLoad (internal/modules/cjs/loader.js:537:12)'
2018-11-28 11:28:21,216 ERROR b'    at Function.Module._load (internal/modules/cjs/loader.js:529:3)'
2018-11-28 11:28:21,216 ERROR b'    at Module.require (internal/modules/cjs/loader.js:636:17)'
2018-11-28 11:28:21,216 ERROR b'    at require (internal/modules/cjs/helpers.js:20:18)'
2018-11-28 11:28:21,216 ERROR b'    at Object.<anonymous> (/home/jin/minos/minos/server/server.js:7:11)'
2018-11-28 11:28:21,216 ERROR b'    at Module._compile (internal/modules/cjs/loader.js:688:30)'
2018-11-28 11:28:21,217 INFO Finished simserver stdout
2018-11-28 11:28:21,218 INFO Finished simserver stderr
2018-11-28 11:28:21,900 INFO Waiting for simserver stdout to finish
2018-11-28 11:28:21,901 INFO Waiting for simserver stderr to finish

ZhuFengdaaa, 2018-11-28:
https://github.com/airbnb/enzyme/issues/942#issuecomment-314715229

romi-seecs, 2018-12-12:
Hi @ZhuFengdaaa @jsdd25 thank you for your help but I don't know JavaScript and need to understand what I have to do with following code:
```

var jsdom = require('jsdom');
const { JSDOM } = jsdom;

const { document } = (new JSDOM('')).window;
global.document = document;
```

Do I have to access home/romi/minos/minos/server/node_modules/sstk/ssc/stk-ssc.js:25
and modify code there? If yes then what do I do after that?

Please help me out here!

angelxuanchang, 2018-12-13:
Thanks @ZhuFengdaaa for identifying the problem!  @romi-seecs You can check which jsdom version you are using by looking at the file minos/server/node_modules/sstk/node_modules/jsdom/package.json 
It should be 9.12.0 - jsdom 11.0.0 and higher will not work. 
I'm not sure why you got a newer version installed.  You can try to remove minos/server/node_modules and run "npm install" again in the minos/server to see if the jsdom version is changed.

jsdd25, 2018-12-13:
@romi-seecs I have fixed this problem. Welcome to get in touch with me. My e-mail is 773903267@qq.com

romi-seecs, 2018-12-18:
Thanks a lot guys @jsdd25 @angelxuanchang , that helped me solve the problem.
But now when I give the command:

```
romi@romi-HP-Compaq-8200-Elite-MT-PC:~/minos$ python3 -m minos.tools.pygame_client
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
2018-12-18 18:27:32,652 INFO {'sim_git_hash': '21dd855', 'machine': 'romi-HP-Compaq-8200-Elite-MT-PC', 'sim_id': 'sim00', 'stk_git_hash': '21dd855'}
Starting simulator...
2018-12-18 18:27:32,669 INFO sim00:Starting sim server at /home/romi/minos/minos/server/server.js with port 47860
Traceback (most recent call last):
  File "/home/romi/minos/minos/tools/pygame_client.py", line 438, in main
    ep_info = sim.start()
  File "/home/romi/minos/minos/lib/Simulator.py", line 321, in start
    started = self.start_child_servers()
  File "/home/romi/minos/minos/lib/Simulator.py", line 255, in start_child_servers
    self._sio = SocketIO(self.params.host, self.params.port)
  File "/home/romi/.local/lib/python3.5/site-packages/socketIO_client/__init__.py", line 345, in __init__
    resource, hurry_interval_in_seconds, **kw)
  File "/home/romi/.local/lib/python3.5/site-packages/socketIO_client/__init__.py", line 58, in __init__
    self._transport
  File "/home/romi/.local/lib/python3.5/site-packages/socketIO_client/__init__.py", line 66, in _transport
    self._engineIO_session = self._get_engineIO_session()
  File "/home/romi/.local/lib/python3.5/site-packages/socketIO_client/__init__.py", line 80, in _get_engineIO_session
    transport.recv_packet())
  File "/home/romi/.local/lib/python3.5/site-packages/socketIO_client/transports.py", line 92, in recv_packet
    for engineIO_packet in decode_engineIO_content(response.content):
  File "/home/romi/.local/lib/python3.5/site-packages/socketIO_client/parsers.py", line 95, in decode_engineIO_content
    content, content_index)
  File "/home/romi/.local/lib/python3.5/site-packages/socketIO_client/parsers.py", line 202, in _read_packet_length
    while get_byte(content, content_index) not in [0, 1]:
  File "/home/romi/.local/lib/python3.5/site-packages/socketIO_client/symmetries.py", line 36, in get_byte
    return six.indexbytes(x, index)
IndexError: index out of range
Error running simulator. Aborting.
2018-12-18 18:27:35,097 INFO sim00:Stopping the simulator
2018-12-18 18:27:35,097 INFO Counter()
2018-12-18 18:27:35,097 INFO sim00:Stopping child servers
2018-12-18 18:27:35,097 INFO sim00:Killing sim pid 3302
2018-12-18 18:27:35,106 INFO sim00:Stopped child servers
2018-12-18 18:27:35,106 INFO sim00:Simulator killed.

```
My log file gives following report:

```
2018-12-18 18:27:33,259 INFO b'Configuration { fileCache: { size: 50 },'
2018-12-18 18:27:33,259 INFO b'  imageCache: { size: 200 },'
2018-12-18 18:27:33,260 INFO b'  imageQueue: { concurrency: 4 },'
2018-12-18 18:27:33,260 INFO b"  base_url: '/home/romi/work/',"
2018-12-18 18:27:33,260 INFO b"  assets_url: 'file:///home/romi/minos/minos/server/node_modules/sstk/ssc/../server/static' }"
2018-12-18 18:27:33,624 INFO b'Waiting for client connection on port 47860'
2018-12-18 18:27:33,643 ERROR b'Cannot register unknown asset 2d3ds'
2018-12-18 18:27:33,643 INFO b'register ../server/static/data/matterport/matterport3d.json'
2018-12-18 18:27:33,646 INFO b'Registered asset group: mp3d'
2018-12-18 18:27:33,650 INFO b'Got 90 assets'
2018-12-18 18:27:33,705 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2018-12-18 18:27:33,706 INFO b'Replace asset group p5d'
2018-12-18 18:27:33,707 INFO b'Registered asset group: p5d'
2018-12-18 18:27:33,791 INFO b'Got 2623 assets'
2018-12-18 18:27:34,704 INFO b'Loaded lightSpecs for 202 models.'
2018-12-18 18:27:34,704 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2018-12-18 18:27:34,707 INFO b'Replace asset group p5dTexture'
2018-12-18 18:27:34,707 INFO b'Registered asset group: p5dTexture'
2018-12-18 18:27:34,944 INFO b'Got 67072 assets'
2018-12-18 18:27:35,097 INFO Waiting for simserver stdout to finish
2018-12-18 18:27:35,105 INFO Finished simserver stdout
2018-12-18 18:27:35,105 INFO Finished simserver stderr
2018-12-18 18:27:35,105 INFO Waiting for simserver stderr to finish
```

I have suncg and matterport3D data sets under home/work folder.
Please help me out here

msavva, 2018-12-18:
@romi-seecs This appears to be an error related to the socket IO package version.  Please try following the instructions here: https://github.com/minosworld/minos/blob/master/FAQ.md#i-get-an-error-from-the-socketio_client-package

romi-seecs, 2018-12-25:
That didn't work however copying mp3d dataset again in work folder worked for me. 
Thanks a lot everyone for your help, you made this so easy for me :)

closed 2018-12-25 by romi-seecs