minosworld/minos #125: MINOS Map issue on matterport

Usama323, 2018-11-23:
Whenever I run command for map then it gives issue for materport data set, It is not giving any error in case of Sunncg data set. Currently I am facing following problem 

python3 -m minos.tools.pygame_client --source mp3d --scene_ids EDJbREhghzL -s map --navmap --task point_goal --width 600 --height 500 --save_png

pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
2018-11-23 17:54:31,338 INFO {'machine': 'usama-HP-Notebook', 'sim_id': 'sim00', 'stk_git_hash': '478f622', 'sim_git_hash': '478f622'}
Starting simulator...
2018-11-23 17:54:31,355 INFO sim00:Starting sim server at /home/usama/Desktop/minosmi/minos/minos/server/server.js with port 29135
2018-11-23 17:54:33,620 INFO sim00:connect
Traceback (most recent call last):
  File "/home/usama/Desktop/minosmi/minos/minos/tools/pygame_client.py", line 438, in main
    ep_info = sim.start()
  File "/home/usama/Desktop/minosmi/minos/minos/lib/Simulator.py", line 327, in start
    self._rpc('start', self.params, self.on_started)
  File "/home/usama/Desktop/minosmi/minos/minos/lib/Simulator.py", line 185, in _rpc
    return rpc.call(name, data, callback, seconds, check_wait=lambda: self.running)
  File "/home/usama/Desktop/minosmi/minos/minos/lib/util/RpcCall.py", line 25, in call
    self.sio.wait_for_callbacks(seconds=seconds)    # wait for response
  File "/home/usama/.local/lib/python3.5/site-packages/socketIO_client/__init__.py", line 450, in wait_for_callbacks
    self.wait(seconds, for_callbacks=True)
  File "/home/usama/.local/lib/python3.5/site-packages/socketIO_client/__init__.py", line 245, in wait
    self._process_packets()
  File "/home/usama/.local/lib/python3.5/site-packages/socketIO_client/__init__.py", line 269, in _process_packets
    self._process_packet(engineIO_packet)
  File "/home/usama/.local/lib/python3.5/site-packages/socketIO_client/__init__.py", line 496, in _process_packet
    delegate(packet)
  File "/home/usama/.local/lib/python3.5/site-packages/socketIO_client/__init__.py", line 523, in _on_ack
    ack_callback(*packet.args)
  File "/home/usama/Desktop/minosmi/minos/minos/lib/util/RpcCall.py", line 92, in _handle_response
    self.result = self.callback(self.response)
  File "/home/usama/Desktop/minosmi/minos/minos/lib/Simulator.py", line 558, in on_started
    self.__process_goal_observations(self.start_summary_info.get('goalObservations'))
  File "/home/usama/Desktop/minosmi/minos/minos/lib/Simulator.py", line 537, in __process_goal_observations
    self.__process_observation(obs)
  File "/home/usama/Desktop/minosmi/minos/minos/lib/Simulator.py", line 509, in __process_observation
    converted = self.__process_camera_frame('map', observation['map'])
  File "/home/usama/Desktop/minosmi/minos/minos/lib/Simulator.py", line 466, in __process_camera_frame
    data = np.reshape(f['data'], f['shape'])
TypeError: list indices must be integers or slices, not str
Error running simulator. Aborting.
2018-11-23 17:55:09,463 INFO sim00:Stopping the simulator
2018-11-23 17:55:09,463 INFO Counter()
2018-11-23 17:55:09,466 INFO sim00:disconnect
2018-11-23 17:55:09,466 INFO sim00:Stopping child servers
2018-11-23 17:55:09,466 INFO sim00:Killing sim pid 2710
2018-11-23 17:55:09,571 INFO sim00:Stopped child servers
2018-11-23 17:55:09,571 INFO sim00:Simulator killed.

ducha-aiki, 2018-11-27:
I had this problem too, but no workaround so far. Try to select another map, somehow some maps are ok

ZhuFengdaaa, 2018-11-28:
```data = np.reshape(f['data'], f['shape'])```

Here f is a list. So use `f=f[0]` get the python dictionary you want. 

It works on most maps.

Usama323, 2018-11-28:
@ZhuFengdaaa thanks , I would try it by changing in which python file ?

ZhuFengdaaa, 2018-11-28:
my modification https://github.com/minosworld/minos/pull/128/commits/5e0ffba91868f6b4db53990ea7df055e15a13c27

Usama323, 2018-11-28:
Thanks @ZhuFengdaaa