minosworld/minos #126: How coud I Print postion of agent

Usama323, 2018-11-23:
I want to print the position and angle of agent , just like nvmap command is providing. If I could save them in excel or text form or could display them on terminal continuously.  
[image]
[image]

Like shown in above images

Usama323, 2018-12-06:
kindly reply me @msavva  or @angelxuanchang  , I really need answer to this question, looking forward for your reply

angelxuanchang, 2018-12-07:
You can use the `--log_action_trace true` option to save `action_trace.csv` in the log directory.  This option should work for both `pygame_client` and the gym demo.  This file will contain the position of the agent in the "px,py,pz" fields.  You can use `minos/server/visualize_traces.js --trace_file <filename>` to visualize the action trace.  

The position is also reported by in the `info` field of the response.  The info field should look something like this:  
```
{'agent_state': 
  {'position': [4.281050989983723, 0.53564888, -3.020889699499121],
   'orientation': [-0.8588654959923366, 0, 0.5122011907384025]}, 
   'goal': [
       {'type': 'position', 'position': [1.2792186244623651, 0.53564888, -1.7029638995910343], 
         'room': ['R5'], 'roomType': ['dining room'], 
          'initialOffsetFromAgent': [2.884750516649895, 0, -2.681519118953451], 
           'cell': {'i': 41, 'j': 99, 'id': 8258, 'isValid': True}
    }]
}
```
See https://github.com/minosworld/minos/blob/master/minos/tools/pygame_client.py#L103 . (you can print out the response by pressing 'o' when using the `pygame_client`)

Usama323, 2018-12-07:
Thanks @angelxuanchang , can you kindly explain the orientation and position ? I mean how they are working ?, Also explain their unit (i.e) cm,mm,um etc for position and rad,degree for angle ? Where is origin of axis in a room ? Every time i start minos and press 'O' it give values in negative for position. Thanks in advance for this act of kindness, looking forward for your reply

angelxuanchang, 2018-12-08:
The scene unit is in meters.  The position is the absolute position of the agent in the scene.  There is no special meaning for the position 0,0,0.  The orientation of the agent is the direction vector for the agent.

Usama323, 2018-12-10:
Ok! Thanks @angelxuanchang . Can you now kindly explain the direction vector (what is x,y,z in this case) little bit, I mean how it is working what does (-0.999) means and what does (0.999) means ? Also can you kindly explain the direction vector along up and down (ceiling and floor), how is that working ? what does -1 means and what does 1 means? Thanks in advance for this act of kindness. As I am trying to get ground truth from MINOS simulator I need to understand these.

Usama323, 2018-12-12:
@angelxuanchang can you kindly reply to my last comment?, I would figure-out the rest myself. But for this I need to understand orientation.

angelxuanchang, 2018-12-13:
The direction vector is a normalized vector in Minos world coordinate frame indicating the direction the agent is facing.  The agent uses a coordinate frame with Y up and -Z front.  Minos uses a world coordinate frame with the same conventions: Y up and -Z front.  Both SUNCG and Matterport3d scenes are rotated to match this (SUNCG from Y up, +Z front and Matterport3D Z up, -X front, the front is somewhat arbitrary).   The Y component of the orientation should be 0 since the agent is just moving in the XZ plane.  If you want the relative orientation of the agent to the goal, you will find that in `observation.measurements.direction_to_goal`

Usama323, 2018-12-13:
ok, and can you tell that "observation.measurements.direction_to_goal" is a command or some file? where to get it ? that's all and thanks for such a explained reply ! :)

angelxuanchang, 2018-12-13:
It is part of the observations that is returned by the simulator at each step.  Please see https://github.com/minosworld/minos/blob/master/minos/tools/pygame_client.py#L84 for example of how the measurements are extracted and https://github.com/minosworld/minos/blob/master/API.md#return-object for example of what is returned.

closed 2018-12-15 by angelxuanchang