minosworld/minos #132: Does MINOS work on Ubuntu 14.04?

romi-seecs, 2018-12-05:
Is it true that MINOS can't be installed on versions older than Ubuntu 16? I installed MINOS on Ubuntu 16 but it won't install on Ubuntu 14.04 LTS

romi-seecs, 2018-12-06:
The problem occurring during installation is when the command pip3 install -e .  is given.
The python modules such as numpy, matplotlib give installation errors.

angelxuanchang, 2018-12-07:
It should work on Ubuntu 14.04 as well.  Please make sure you have system dependencies installed.   If you have errors, please provide more information.

romi-seecs, 2018-12-13:
@angelxuanchang Thank you for your response! npm doesn't install inside server directory and following errors occur. I hope you can help me out here, P.S. I've never got such error in Ubuntu 16.

```

ayyaz@ubuntu:~/minos/minos/server$ npm install

> sstk@0.5.3 preinstall /home/ayyaz/minos/minos/server/node_modules/sstk
> ./getdata.sh

Cloning into './metadata'...
remote: Enumerating objects: 29, done.
remote: Counting objects: 100% (29/29), done.
remote: Compressing objects: 100% (18/18), done.
remote: Total 94 (delta 23), reused 11 (delta 11), pack-reused 65
Unpacking objects: 100% (94/94), done.
Checking connectivity... done.

> node-expat@2.3.16 install /home/ayyaz/minos/minos/server/node_modules/node-expat
> node-gyp rebuild

make: Entering directory
`/home/ayyaz/minos/minos/server/node_modules/node-expat/build'
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmlparse.o
../deps/libexpat/lib/xmlparse.c: In function ‘gather_time_entropy’:
../deps/libexpat/lib/xmlparse.c:780:7: warning: variable
‘gettimeofday_res’ set but not used [-Wunused-but-set-variable]
   int gettimeofday_res;
       ^
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmltok.o
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmlrole.o
  AR(target) Release/obj.target/deps/libexpat/libexpat.a
  COPY Release/libexpat.a
  CXX(target) Release/obj.target/node_expat/node-expat.o
make: g++: Command not found
make: *** [Release/obj.target/node_expat/node-expat.o] Error 127
make: Leaving directory
`/home/ayyaz/minos/minos/server/node_modules/node-expat/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit
(/home/ayyaz/.nvm/versions/node/v10.14.2/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit
(internal/child_process.js:240:12)
gyp ERR! System Linux 3.13.0-32-generic
gyp ERR! command "/home/ayyaz/.nvm/versions/node/v10.14.2/bin/node"
"/home/ayyaz/.nvm/versions/node/v10.14.2/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js"
"rebuild"
gyp ERR! cwd /home/ayyaz/minos/minos/server/node_modules/node-expat
gyp ERR! node -v v10.14.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-expat@2.3.16 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-expat@2.3.16 install script.
npm ERR! This is probably not a problem with npm. There is likely
additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ayyaz/.npm/_logs/2018-12-13T18_03_58_706Z-debug.log
```

angelxuanchang, 2018-12-13:
You are missing the g++ compiler.  You can install it using: `sudo apt-get install build-essential`
It's possible you are missing some other packages on ubuntu 14.04 as well: 
try `sudo apt-get install build-essential libxi-dev libglu1-mesa-dev libglew-dev`

romi-seecs, 2018-12-16:
Oops! Yes you're right that helped, but now when I try to install the minos Python module by running 'pip3 install -e .' in the root of the repository, pygame gives following installation error:

Unable to run "sdl-config". Please make sure a development version of SDL is installed.

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip_build_ayyaz/pygame
Storing debug log for failure in /home/ayyaz/.pip/pip.log

I tried installing manually by 'pip3 install pygame' but got same error.
Can you help me out here?

romi-seecs, 2018-12-16:
Now that I've installed every dependency and Python module, I'm getting this issue

ayyaz@ubuntu:~/minos$ python3 -m minos.tools.pygame_client --empty_room 
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ayyaz/minos/minos/tools/pygame_client.py", line 4, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'

Despite the fact that all modules have been installed

ayyaz@ubuntu:~/minos$ pip3 install -e .
Obtaining file:///home/ayyaz/minos
  Running setup.py (path:/home/ayyaz/minos/setup.py) egg_info for package from file:///home/ayyaz/minos
    
Requirement already satisfied (use --upgrade to upgrade): socketIO-client-2 in /usr/local/lib/python3.4/dist-packages (from minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): numpy in /usr/lib/python3/dist-packages (from minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): scipy in /usr/local/lib/python3.4/dist-packages (from minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): easydict in /usr/local/lib/python3.4/dist-packages (from minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): matplotlib in /usr/lib/python3/dist-packages (from minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): pygame in /home/ayyaz/.local/lib/python3.4/site-packages (from minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): Pillow in /usr/lib/python3/dist-packages (from minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): psutil in /usr/local/lib/python3.4/dist-packages (from minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): pyyaml in /usr/local/lib/python3.4/dist-packages (from minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): requests in /usr/lib/python3/dist-packages (from socketIO-client-2->minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): six in /usr/lib/python3/dist-packages (from socketIO-client-2->minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): websocket-client in /usr/local/lib/python3.4/dist-packages (from socketIO-client-2->minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): python-dateutil in /usr/lib/python3/dist-packages (from matplotlib->minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): tornado in /usr/lib/python3/dist-packages (from matplotlib->minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): pyparsing>=1.5.6 in /usr/lib/python3/dist-packages (from matplotlib->minos-sim==0.5.3)
Requirement already satisfied (use --upgrade to upgrade): nose in /usr/lib/python3/dist-packages (from matplotlib->minos-sim==0.5.3)
Installing collected packages: minos-sim
  Running setup.py develop for minos-sim
    Checking .pth file support in /usr/local/lib/python3.4/dist-packages/
    /usr/bin/python3 -E -c pass
    TEST PASSED: /usr/local/lib/python3.4/dist-packages/ appears to support .pth files
    
    Creating /usr/local/lib/python3.4/dist-packages/minos-sim.egg-link (link to .)
    minos-sim 0.5.3 is already the active version in easy-install.pth
    
    Installed /home/ayyaz/minos
Successfully installed minos-sim
Cleaning up...

Please help me out here

angelxuanchang, 2018-12-17:
It looks like you have both /usr/local/lib/python3.6 and /usr/local/lib/python3.4  and that pip3 is installing into /usr/lib/python3 and  /usr/local/lib/python3.4 while you are running from /usr/local/lib/python3.6

closed 2019-07-07 by romi-seecs