minosworld/minos #133: What does rotation means in action_trace file ?

Usama323, 2018-12-13:
When I save action_trace file, there is column of rotation, what does it means? the rotation is in degrees or radians ? Looking forward for the reply !

DRAhmadFaraz, 2018-12-13:
I am also facing the same issues. "reply if you have solved this issue".

angelxuanchang, 2018-12-13:
The rotation is the orientation of the agent encoded as a single angle in radius around the up axis (Y).

Usama323, 2018-12-13:
radius is unit of angle ? or it is something else ?

angelxuanchang, 2018-12-13:
sorry, radians

Usama323, 2018-12-13:
Thanks @angelxuanchang for fast reply !

Usama323, 2018-12-14:
One last question @angelxuanchang, that is this angle is in range of “-pi to pi” or “-2pi to 2pi” ?

Usama323, 2018-12-16:
Thanks a lot for the response

closed 2018-12-16 by angelxuanchang

Usama323, 2019-04-26:
Hi @angelxuanchang  I have seen value of rotation in action_trace  files and it variate for different files. How can one generalize it ? 
What is its range as in some files value of rotation is -10.8471 and in some files it is 0.4913. 
Also how to convert it in degree , if I use the well known formula to convert from radian to degree (1rad * 180/pi) then it gives value more than 360. 

Kindly reply me as soon as possible, looking forward for your reply. Thanks in advance for this act of kindness.