minosworld/minos #135: Evaluation does not follow the episode schedule in point_goal task / goal position not passed.

mcimpoi, 2018-12-16:
Hi, 

I tried several runs of the same agent (unreal), in the attempt to reproduce the results in Table2 in the paper, namely PoingGoal for Matterport3D small and medium.

However, looking at the logs, filtered by "EPISODE", for the first episode, at test time, the goal is at different coordinates (* * was added manually ):
```
{
  'sceneId': 'mp3d.2t7WUuJeko7',
  'task': 'point_goal',
  'start': {
    'angle': 1.386,
    'position': [0.672, 0.551, 2.339],
    'cell': {
      'j': 39,
      'isValid': True,
      'i': 80,
      'id': 3707
    }
  },
  'goal': {
    'angle': 1.8403473282415732,

    **'position': [-4.813260249729765, 0.5735506993532181, 1.2931640217578368],**

    'room': 'R5',
    'cell': {
      'j': 29,
      'isValid': True,
      'i': 25,
      'id': 2722
    },
    'radius': 0.25
  },
  'shortestPath': {
    'distance': 7.601219330881972,
    'isValid': True,
    'doors': [],
    'rooms': [1, 5, 6]
  }
}

{
  'start': {
    'position': [0.672, 0.551, 2.339],
    'cell': {
      'id': 3707,
      'j': 39,
      'i': 80,
      'isValid': True
    },
    'angle': 1.386
  },
  'task': 'point_goal',
  'sceneId': 'mp3d.2t7WUuJeko7',
  'goal': {
    'radius': 0.25,

    **'position': [-1.235677195070037, 0.5514386199228466, -0.9918128308026164],**

    'cell': {
      'id': 619,
      'j': 6,
      'i': 61,
      'isValid': True
    },
    'room': 'R0',
    'angle': 0.08224559970336502
  },
  'shortestPath': {
    'distance': 4.262741699796953,
    'doors': [],
    'isValid': True,
    'rooms': [1]
  }
}
```

The goal coordinates are different for different runs, and also different from what the episode scheduler for test is providing: 

```
{
  "pathNumDoors": 0,
  "pathNumRooms": 3,
  "episode_id": 10714,
  "level": 0,
  "pathDoorIds": "",
  "dist": 5.209,
  "pathDist": 6.36,
  "goal": {
    "position": [-4.473, 0.029, 1.72],
    "id": ""
  },
  "scene_id": "2t7WUuJeko7",
  "room_id": "R5",
  "start": {
    "position": [0.672, 0.551, 2.339],
    "angle": 1.386
  }
}
```
Looking in RoomSimulator.py, seems the problem is in L96-103
```
        if 'goal' in ep_settings:
            if self.params['task'] == 'room_goal':
                room_id = ep_settings['room_id']
                config['goal'] = {'roomIds': [room_id]}
            else:
                object_id = ep_settings['goal']['id']
                if object_id:
                    config['goal'] = {'objectIds': [object_id]}
```

Which doesn't handle the point_goal case.
If I try to set config['goal'] I get the simulator crash with the following stack trace:

```
 20:36:41,008 INFO b'set active level 0'
 20:36:41,015 ERROR b"Error fetching model-mp3d.2t7WUuJeko7-matterport-glb TypeError: Cannot read property 'position' of undefined"
 20:36:41,016 ERROR b'    at NavScene.getCellId (/research/minos_world_forks/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:101809:42)'
 20:36:41,016 ERROR b'    at /research/minos_world_forks/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:101988:21'
 20:36:41,016 ERROR b'    at arrayMap (/research/minos_world_forks/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:4651:24)'
 20:36:41,016 ERROR b'    at Function.map (/research/minos_world_forks/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:13568:15)'
 20:36:41,016 ERROR b'    at NavScene.computeShortestPath (/research/minos_world_forks/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:101987:24)'
 20:36:41,016 ERROR b'    at NavScene.reset (/research/minos_world_forks/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:101679:31)'
 20:36:41,016 ERROR b'    at SimState.reset (/research/minos_world_forks/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:111700:24)'
 20:36:41,016 ERROR b'    at Simulator.__setSceneState (/research/minos_world_forks/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:110007:15)'
 20:36:41,016 ERROR b'    at /research/minos_world_forks/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:110243:12'
 20:36:41,017 ERROR b'    at wrappedCallback (/research/minos_world_forks/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:30066:10)'
 20:36:41,017 ERROR b"Cannot load: TypeError: Cannot read property 'position' of undefined"
```

Any hints in how to fix this? 

Thank you!

msavva, 2018-12-18:
Hi @mcimpoi ,
Thank you for trying out MINOS and for reporting this bug.  We have committed a fix to the master branch which should correctly handle `point_goal` and make goal specification handling a bit more robust overall.  Please try running with the fix, and let us know if it the issue is resolved.

angelxuanchang, 2018-12-24:
Thanks for letting us know about this - we have fixed `override_args` to default to `{}` in `sim_config.py`
I'm closing this issue now - please let us know if you encounter other issues.

closed 2018-12-24 by angelxuanchang

mcimpoi, 2019-01-11:
Hi @angelxuanchang, @msavva,

I ran a simple test with the following code snippet -- trying to debug the same problem, and still did not get it running.

Here is what I tried (running version 9f3954207aeccabdc2df283b3706092e4466a9e9 / sstk@0.6.0 (github:smartscenes/sstk#3735e535dd4e7d7b006c197e14e9bdef356f28c8) ):

`
from minos.config.sim_args import sim_config
import minos.lib.common as common
from minos.lib.Simulator import Simulator

simargs = sim_config.get('pointgoal_mp3d_s')
sim = Simulator(simargs)
sim.init()

# get the first test episode
curr_schedule = 'test'
ep_schedulers = common.create_episode_schedulers(simargs)
test_ep_scheduler = ep_schedulers['test']
test_ep_scheduler.reset()
ep_settings = test_ep_scheduler.next_episode()

#create config
config = {}
config['scene'] = {'fullId': simargs['scene']['dataset'] + '.' + ep_settings['scene_id'],
                               'level': ep_settings['level'],
                               'textureSet': curr_schedule}
config['goal'] =  {'type': 'position', 'position': ep_settings['goal']['position']}
config['start'] = ep_settings['start']

sim.configure(config)
result = sim.start()  # And here are the problems :)
`

I get lots of  `WARNING localhost:50319/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=50319): Max retries exceeded with url: /socket.io/?t=1547218508808-0&EIO=3&transport=polling (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fdd7a7d9550>: Failed to establish a new connection: [Errno 111] Connection refused',))`

And in the logs I get this: 
`
15:40:49,353 INFO b'Loading scene  mp3d.gxdoqLR6rwA'
15:40:49,358 ERROR b'/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:108253'
15:40:49,358 ERROR b'\t  if (this.opts.scene && this.opts.scene.fullId.startsWith("mp3d")) {'
15:40:49,358 ERROR b'\t                                                ^'
15:40:49,358 ERROR b''
15:40:49,358 ERROR b"TypeError: Cannot read property 'startsWith' of undefined"
15:40:49,358 ERROR b'    at Simulator.__loadScene (/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:108253:50)'
15:40:49,358 ERROR b'    at /minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:108205:14'
15:40:49,358 ERROR b'    at /minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:108126:8'
15:40:49,358 ERROR b'    at /minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:39080:17'
15:40:49,358 ERROR b'    at next (/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:43936:30)'
15:40:49,358 ERROR b'    at /minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:39576:17'
15:40:49,358 ERROR b'    at Timeout._onTimeout (/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:108122:35)'
15:40:49,358 ERROR b'    at ontimeout (timers.js:436:11)'
15:40:49,359 ERROR b'    at tryOnTimeout (timers.js:300:5)'
15:40:49,359 ERROR b'    at listOnTimeout (timers.js:263:5)'
15:40:49,370 INFO Finished simserver stderr
15:40:49,371 INFO Finished simserver stdout
`

Many thanks once again.

PS: Is there an easy way to debug SSTK related errors? (e.g. using the "source" file, not the generated code? sstk/client/build/STK-core.bundle.js:108122:35)

angelxuanchang, 2019-02-01:
Thanks for reporting this issue.  This should be now fixed.