minosworld/minos #141: Socket IO issue

DRAhmadFaraz, 2018-12-20:
I have installed all the dependencies successfully but I have doubt on "socketIO_client" for this version of Minos.
 
here is my logs/ simserver.logs file status. kindly guide me what to do...

2018-12-20 22:15:44,843 INFO b'Configuration { fileCache: { size: 50 },'
2018-12-20 22:15:44,843 INFO b'  imageCache: { size: 200 },'
2018-12-20 22:15:44,844 INFO b'  imageQueue: { concurrency: 4 },'
2018-12-20 22:15:44,844 INFO b"  base_url: '/home/ahmad/work/',"
2018-12-20 22:15:44,844 INFO b'  assets_url:'
2018-12-20 22:15:44,844 INFO b"   'file:///home/ahmad/Desktop/minos/minos/server/node_modules/sstk/ssc/../server/static' }"
2018-12-20 22:15:45,068 INFO b'Waiting for client connection on port 39953'
2018-12-20 22:15:45,085 ERROR b'Cannot register unknown asset 2d3ds'
2018-12-20 22:15:45,086 INFO b'register ../server/static/data/matterport/matterport3d.json'
2018-12-20 22:15:45,088 INFO b'Registered asset group: mp3d'
2018-12-20 22:15:45,092 INFO b'Got 90 assets'
2018-12-20 22:15:45,166 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2018-12-20 22:15:45,168 INFO b'Replace asset group p5d'
2018-12-20 22:15:45,168 INFO b'Registered asset group: p5d'
2018-12-20 22:15:45,246 INFO b'Got 2623 assets'
2018-12-20 22:15:46,078 INFO b'Loaded lightSpecs for 202 models.'
2018-12-20 22:15:46,079 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2018-12-20 22:15:46,083 INFO b'Replace asset group p5dTexture'
2018-12-20 22:15:46,083 INFO b'Registered asset group: p5dTexture'
2018-12-20 22:15:46,342 INFO b'Got 67072 assets'
2018-12-20 22:15:46,507 INFO b'Client GdNoxJjZre6SIqsXAAAA connected on port 39953'
2018-12-20 22:15:46,581 INFO b'THREE.WebGLRenderer 84'
2018-12-20 22:15:46,582 ERROR b'THREE.WebGLRenderer: WEBGL_depth_texture extension not supported.'
2018-12-20 22:15:46,582 ERROR b'THREE.WebGLRenderer: OES_texture_float_linear extension not supported.'
2018-12-20 22:15:46,582 ERROR b'THREE.WebGLRenderer: OES_texture_half_float extension not supported.'
2018-12-20 22:15:46,582 ERROR b'THREE.WebGLRenderer: OES_texture_half_float_linear extension not supported.'
2018-12-20 22:15:46,582 ERROR b'THREE.WebGLRenderer: OES_standard_derivatives extension not supported.'
2018-12-20 22:15:46,593 INFO b'Creating Simulator with options { frameSkip: 1,'
2018-12-20 22:15:46,594 INFO b'  episodesPerSceneTest: 1,'
2018-12-20 22:15:46,594 INFO b"  rewardType: 'dist_time',"
2018-12-20 22:15:46,594 INFO b'  roomtypesFile:'
2018-12-20 22:15:46,594 INFO b"   '/home/ahmad/Desktop/minos/minos/config/../data/roomTypes.suncg.csv',"
2018-12-20 22:15:46,594 INFO b'  sensors:'
2018-12-20 22:15:46,594 INFO b"   [ { name: 'group',"
2018-12-20 22:15:46,594 INFO b"       type: 'group',"
2018-12-20 22:15:46,594 INFO b'       orientation: [Array],'
2018-12-20 22:15:46,594 INFO b'       position: [Array],'
2018-12-20 22:15:46,594 INFO b'       fov: 45,'
2018-12-20 22:15:46,594 INFO b'       active: true,'
2018-12-20 22:15:46,594 INFO b'       far: 20,'
2018-12-20 22:15:46,594 INFO b'       resolution: [Array],'
2018-12-20 22:15:46,594 INFO b'       modes: [Array],'
2018-12-20 22:15:46,594 INFO b"       configuration: 'positional',"
2018-12-20 22:15:46,594 INFO b'       near: 0.001,'
2018-12-20 22:15:46,594 INFO b'       resize: true },'
2018-12-20 22:15:46,594 INFO b'     { noise_model: [Object],'
2018-12-20 22:15:46,594 INFO b"       encoding: 'depth',"
2018-12-20 22:15:46,594 INFO b"       name: 'depth',"
2018-12-20 22:15:46,594 INFO b"       type: 'depth',"
2018-12-20 22:15:46,594 INFO b'       orientation: [Array],'
2018-12-20 22:15:46,594 INFO b'       noise: false,'
2018-12-20 22:15:46,594 INFO b'       position: [Array],'
2018-12-20 22:15:46,595 INFO b'       fov: 45,'
2018-12-20 22:15:46,595 INFO b'       active: true,'
2018-12-20 22:15:46,595 INFO b'       far: 20,'
2018-12-20 22:15:46,595 INFO b'       resolution: [Array],'
2018-12-20 22:15:46,595 INFO b"       datatype: 'float32',"
2018-12-20 22:15:46,595 INFO b"       configuration: 'positional',"
2018-12-20 22:15:46,595 INFO b'       metersToUnit: 1,'
2018-12-20 22:15:46,595 INFO b'       near: 0.1,'
2018-12-20 22:15:46,595 INFO b'       resize: true },'
2018-12-20 22:15:46,595 INFO b"     { encoding: 'pcm',"
2018-12-20 22:15:46,595 INFO b"       name: 'audio',"
2018-12-20 22:15:46,595 INFO b"       type: 'audio',"
2018-12-20 22:15:46,595 INFO b'       orientation: [Array],'
2018-12-20 22:15:46,595 INFO b'       position: [Array],'
[... 306 lines cut ...]
2018-12-20 22:15:46,638 ERROR b"  code: 'ENOENT',"
2018-12-20 22:15:46,638 ERROR b"  syscall: 'open',"
2018-12-20 22:15:46,638 ERROR b"  path: '/home/ahmad/work//suncg//stats/v4/suncg.materials.csv' }"
2018-12-20 22:15:46,638 INFO b'Removing subscribers for event ready [ 0 ]'
2018-12-20 22:15:46,639 INFO b'importMaterialCsvs: 2.868ms'
2018-12-20 22:15:46,639 INFO b'imported relation indices'
2018-12-20 22:15:46,639 INFO b'importRelationCsvs: 2.713ms'
2018-12-20 22:15:46,640 INFO b'Loading scene  p5dScene.00a76592d5cc7d92eef022393784a2de'
2018-12-20 22:15:46,644 INFO b'Loaded LightSpecs for 202 models'
2018-12-20 22:15:46,648 INFO b'preloading /home/ahmad/work//suncg//wall/00a76592d5cc7d92eef022393784a2de/00a76592d5cc7d92eef022393784a2de.arch.json'
2018-12-20 22:15:46,650 INFO b'Removing subscribers for event drain [ 0 ]'
2018-12-20 22:15:47,848 INFO b'Timing loadScene: 1207.081ms'
2018-12-20 22:15:47,872 INFO b'Loaded p5dScene.00a76592d5cc7d92eef022393784a2de bbdims: [9.500002294778824,3.2560323100497186,6.06973572075367]'
2018-12-20 22:15:47,876 INFO b'Retexture scene'
2018-12-20 22:15:47,976 INFO b'buildBVH.scene: 77.299ms'
2018-12-20 22:15:47,977 INFO b'SimState.sample: 84.030ms'
2018-12-20 22:15:47,981 INFO b'started... waiting for all images to load'
2018-12-20 22:15:48,801 INFO b'Removing subscribers for event drain [ 0 ]'
2018-12-20 22:15:48,804 INFO b'Timing waitImages: 822.276ms'
2018-12-20 22:15:49,004 ERROR b'THREE.WebGLRenderer: EXT_texture_filter_anisotropic extension not supported.'
2018-12-20 22:15:49,164 INFO b'Timing start: 2531.394ms'
2018-12-20 22:15:49,172 INFO b'Received close signal. Shutting down simulator server...'
2018-12-20 22:15:49,173 INFO b'Closed simulator'
2018-12-20 22:15:58,206 INFO b'Received close signal. Shutting down simulator server...'
2018-12-20 22:15:58,206 INFO b'Closed simulator'

DRAhmadFaraz, 2018-12-21:
2018-12-20 22:15:49,004 ERROR b'THREE.WebGLRenderer: EXT_texture_filter_anisotropic extension not supported.'

mcimpoi, 2019-01-16:
Hi! I think you can safely ignore the WebGLRenderer error.

However, checking whether the data is in the correct location might fix your problem (file not found):

> 2018-12-20 22:15:46,638 ERROR b"Error loading scene statistics { [Error: ENOENT: no such file or directory, open '/home/ahmad/work//suncg//stats/v4/suncg.materials.csv']" 

Hope this helps.