minosworld/minos #144: Errors with npm install

leah0206, 2018-12-24:
Hello, I wanted to try minos but after running `npm install` in the server directory got the next errors:
```
ccs@ccs-MS-7996:~/minos-0.6.0/minos/server$ npm install

> sstk@github:smartscenes/sstk#3735e535dd4e7d7b006c197e14e9bdef356f28c8 postinstall /home/ccs/minos-0.6.0/minos/server/node_modules/sstk
> node getdata.js && npm run bootstrap

正克隆到 'metadata'...

> sstk@0.6.0 bootstrap /home/ccs/minos-0.6.0/minos/server/node_modules/sstk
> lerna bootstrap --hoist

lerna info version 2.11.0
lerna info Bootstrapping 4 packages
lerna info lifecycle preinstall
lerna info Installing external dependencies
lerna info hoist Installing hoisted dependencies into root
lerna info hoist Pruning hoisted dependencies
lerna info hoist Finished pruning hoisted dependencies
lerna ERR! execute callback with error
lerna ERR! Error: Command failed: npm install
lerna ERR! npm WARN deprecated filesaverjs@0.2.2: use 'file-saver' instead
lerna ERR! npm WARN deprecated jade@1.11.0: Jade has been renamed to pug, please install the latest version of pug instead of jade
lerna ERR! npm WARN deprecated coffee-script@1.12.7: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
lerna ERR! npm WARN deprecated jscs-preset-wikimedia@1.0.1: Deprecated in favour of eslint-config-wikimedia. -- https://phabricator.wikimedia.org/T118941
lerna ERR! npm WARN deprecated constantinople@3.0.2: Please update to at least constantinople 3.1.1
lerna ERR! npm WARN deprecated transformers@2.1.0: Deprecated, use jstransformer
lerna ERR! npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install the latest version of pug instead of jade
lerna ERR! npm WARN deprecated to-iso-string@0.0.2: to-iso-string has been deprecated, use @segment/to-iso-string instead.
lerna ERR! npm WARN deprecated nomnom@1.8.1: Package no longer supported. Contact support@npmjs.com for more info.
lerna ERR! npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
lerna ERR! npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
lerna ERR! npm WARN deprecated minimatch@2.0.10: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
lerna ERR! npm WARN deprecated hoek@5.0.4: This version is no longer maintained. Please upgrade to the latest version.
lerna ERR! npm WARN deprecated hoek@0.7.6: The major version is no longer supported. Please update to 4.x or newer
lerna ERR! ../deps/libexpat/lib/xmlparse.c: In function ‘gather_time_entropy’:
lerna ERR! ../deps/libexpat/lib/xmlparse.c:780:7: warning: variable ‘gettimeofday_res’ set but not used [-Wunused-but-set-variable]
lerna ERR!    int gettimeofday_res;
lerna ERR!        ^
lerna ERR! In file included from ../../nan/nan_new.h:189:0,
lerna ERR!                  from ../../nan/nan.h:222,
lerna ERR!                  from ../node-expat.cc:1:
lerna ERR! ../../nan/nan_implementation_12_inl.h: In static member function ‘static Nan::imp::FactoryBase<v8::StringObject>::return_t Nan::imp::Factory<v8::StringObject>::New(v8::Local<v8::String>)’:
lerna ERR! ../../nan/nan_implementation_12_inl.h:340:37: warning: ‘static v8::Local<v8::Value> v8::StringObject::New(v8::Local<v8::String>)’ is deprecated: Use Isolate* version [-Wdeprecated-declarations]
lerna ERR!    return v8::StringObject::New(value).As<v8::StringObject>();
lerna ERR!                                      ^
lerna ERR! In file included from /home/ccs/.node-gyp/10.13.0/include/node/v8.h:26:0,
lerna ERR!                  from /home/ccs/.node-gyp/10.13.0/include/node/node.h:63,
lerna ERR!                  from ../../nan/nan.h:52,
lerna ERR!                  from ../node-expat.cc:1:
lerna ERR! /home/ccs/.node-gyp/10.13.0/include/node/v8.h:5053:37: note: declared here
lerna ERR!                  static Local<Value> New(Local<String> value));
lerna ERR!                                      ^
lerna ERR! /home/ccs/.node-gyp/10.13.0/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
lerna ERR!    declarator __attribute__((deprecated(message)))
lerna ERR!    ^
lerna ERR! In file included from ../node-expat.cc:1:0:
lerna ERR! ../../nan/nan.h: In constructor ‘Nan::Utf8String::Utf8String(v8::Local<v8::Value>)’:
lerna ERR! ../../nan/nan.h:1066:53: warning: ‘v8::Local<v8::String> v8::Value::ToString() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
lerna ERR!        v8::Local<v8::String> string = from->ToString();
lerna ERR!                                                      ^
lerna ERR! In file included from /home/ccs/.node-gyp/10.13.0/include/node/node.h:63:0,
lerna ERR!                  from ../../nan/nan.h:52,
lerna ERR!                  from ../node-expat.cc:1:
lerna ERR! /home/ccs/.node-gyp/10.13.0/include/node/v8.h:10040:15: note: declared here
lerna ERR!  Local<String> Value::ToString() const {
lerna ERR!                ^
lerna ERR! In file included from ../node-expat.cc:1:0:
lerna ERR! ../../nan/nan.h:1080:74: warning: ‘int v8::String::WriteUtf8(char*, int, int*, int) const’ is deprecated: Use Isolate* version [-Wdeprecated-declarations]
lerna ERR!          length_ = string->WriteUtf8(str_, static_cast<int>(len), 0, flags);
[... 418 lines cut ...]
  timedOut: false,
  exitCode: 1 }
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! sstk@0.6.0 bootstrap: `lerna bootstrap --hoist`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the sstk@0.6.0 bootstrap script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ccs/.npm/_logs/2018-12-24T06_32_54_880Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! sstk@github:smartscenes/sstk#3735e535dd4e7d7b006c197e14e9bdef356f28c8 postinstall: `node getdata.js && npm run bootstrap`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the sstk@github:smartscenes/sstk#3735e535dd4e7d7b006c197e14e9bdef356f28c8 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ccs/.npm/_logs/2018-12-24T06_32_55_986Z-debug.log
```
Please help me out here!
Thank you very much!

leah0206, 2018-12-25:
Thank you very much! I have successfully run `npm install` with your help!

closed 2018-12-25 by leah0206

TopCoder2K, 2021-11-24:
@leah0206, could you please tell me whether just running `sudo apt-get install build-essential libxi-dev libglu1-mesa-dev libglew-dev libvips` helped you or did you do anything else? I have the same error but I've already installed the libraries before running `npm install` :(((