minosworld/minos #145: Errors with python3 -m minos.tools.pygame_client

leah0206, 2018-12-25:
Hello, I wanted to try minos **v0.6.0**. However, after running `python3 -m minos.tools.pygame_client`, I have got the next errors:
```
ccs@ccs-MS-7996:~/minos-0.6.0$ python3 -m minos.tools.pygame_client
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
Traceback (most recent call last):
  File "/home/ccs/anaconda3/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/ccs/anaconda3/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/ccs/minos-0.6.0/minos/tools/pygame_client.py", line 457, in <module>
    main()
  File "/home/ccs/minos-0.6.0/minos/tools/pygame_client.py", line 415, in main
    sim = Simulator(vars(args))
  File "/home/ccs/minos-0.6.0/minos/lib/Simulator.py", line 110, in __init__
    cwd=stk_sim_path).rstrip(),
  File "/home/ccs/anaconda3/lib/python3.6/subprocess.py", line 336, in check_output
    **kwargs).stdout
  File "/home/ccs/anaconda3/lib/python3.6/subprocess.py", line 418, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['git', 'rev-parse', '--short', 'HEAD']' returned non-zero exit status 128.
2018-12-25 09:58:07,506 INFO sim00:Stopping the simulator
2018-12-25 09:58:07,506 INFO Counter()
2018-12-25 09:58:07,507 INFO sim00:Stopping child servers
2018-12-25 09:58:07,507 INFO sim00:Stopped child servers
2018-12-25 09:58:07,507 INFO sim00:Simulator killed.
```
Please give me some help!
Thank you very much!

angelxuanchang, 2018-12-25:
Thanks you for trying out Minos.   For debugging, we log the git hash of the minos version that is being used.  You will get this error if you are not running the code from an git checkout.

To avoid this, you can try the v0.7.x branch of minos.  If you are using v0.6.0, you can comment out lines 108-112 and 116-117 of https://github.com/minosworld/minos/blob/master/minos/lib/Simulator.py

leah0206, 2018-12-25:
Thank you very much! I have fixed this problem with your help!

closed 2018-12-25 by leah0206