minosworld/minos #147: Getting errors with python3 -m minos.tools.pygame_client

leah0206, 2018-12-25:
Hello, I wanted to try to run `python3 -m minos.tools.pygame_client`. However, I encountered the following problem:
```
ccs@ccs-MS-7996:~/minos-0.6.0$ python3 -m minos.tools.pygame_client
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
2018-12-25 16:01:13,500 INFO {'sim_id': 'sim00', 'machine': 'ccs-MS-7996', 'sim_git_hash': '', 'stk_git_hash': ''}
Starting simulator...
2018-12-25 16:01:13,511 INFO sim00:Starting sim server at /home/ccs/minos-0.6.0/minos/server/server.js with port 48784
2018-12-25 16:01:14,516 INFO sim00:sim server has exited with rv 1
2018-12-25 16:01:14,524 WARNING localhost:48784/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=48784): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1545724874518-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f6ef5193a58>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-12-25 16:01:15,528 WARNING localhost:48784/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=48784): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1545724875525-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f6ef5193fd0>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-12-25 16:01:16,532 WARNING localhost:48784/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=48784): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1545724876529-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f6ef5196358>: Failed to establish a new connection: [Errno 111] Connection refused',))
^CTraceback (most recent call last):
  File "/home/ccs/msavva-socketIO-client-2-bed48c2/socketIO_client/logs.py", line 23, in _yield_warning_screen
    yield elapsed_time
Exception: [waiting for connection] HTTPConnectionPool(host='localhost', port=48784): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1545724876529-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f6ef5196358>: Failed to establish a new connection: [Errno 111] Connection refused',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ccs/minos-0.6.0/minos/tools/pygame_client.py", line 438, in main
    ep_info = sim.start()
  File "/home/ccs/minos-0.6.0/minos/lib/Simulator.py", line 333, in start
    started = self.start_child_servers()
  File "/home/ccs/minos-0.6.0/minos/lib/Simulator.py", line 267, in start_child_servers
    self._sio = SocketIO(self.params.host, self.params.port)
  File "/home/ccs/msavva-socketIO-client-2-bed48c2/socketIO_client/__init__.py", line 345, in __init__
    resource, hurry_interval_in_seconds, **kw)
  File "/home/ccs/msavva-socketIO-client-2-bed48c2/socketIO_client/__init__.py", line 58, in __init__
    self._transport
  File "/home/ccs/msavva-socketIO-client-2-bed48c2/socketIO_client/__init__.py", line 66, in _transport
    self._engineIO_session = self._get_engineIO_session()
  File "/home/ccs/msavva-socketIO-client-2-bed48c2/socketIO_client/__init__.py", line 86, in _get_engineIO_session
    warning_screen.throw(warning)
  File "/home/ccs/msavva-socketIO-client-2-bed48c2/socketIO_client/logs.py", line 29, in _yield_warning_screen
    time.sleep(1)
KeyboardInterrupt
Error running simulator. Aborting.
2018-12-25 16:01:16,631 INFO sim00:Stopping the simulator
2018-12-25 16:01:16,631 INFO Counter()
2018-12-25 16:01:16,632 INFO sim00:Stopping child servers
2018-12-25 16:01:16,632 INFO sim00:Stopped child servers
2018-12-25 16:01:16,632 INFO sim00:Simulator killed.
```
Then I checked the log.
**simserver.log**
```
2018-12-25 16:01:13,599 ERROR b'internal/modules/cjs/loader.js:582'
2018-12-25 16:01:13,600 ERROR b'    throw err;'
2018-12-25 16:01:13,600 ERROR b'    ^'
2018-12-25 16:01:13,600 ERROR b''
2018-12-25 16:01:13,600 ERROR b"Error: Cannot find module 'deasync'"
2018-12-25 16:01:13,600 ERROR b'    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:580:15)'
2018-12-25 16:01:13,600 ERROR b'    at Function.Module._load (internal/modules/cjs/loader.js:506:25)'
2018-12-25 16:01:13,600 ERROR b'    at Module.require (internal/modules/cjs/loader.js:636:17)'
2018-12-25 16:01:13,600 ERROR b'    at require (internal/modules/cjs/helpers.js:20:18)'
2018-12-25 16:01:13,600 ERROR b'    at Object.<anonymous> (/home/ccs/minos-0.6.0/minos/server/node_modules/sstk/ssc/stk-ssc.js:2:15)'
2018-12-25 16:01:13,600 ERROR b'    at Module._compile (internal/modules/cjs/loader.js:688:30)'
2018-12-25 16:01:13,600 ERROR b'    at Object.Module._extensions..js (internal/modules/cjs/loader.js:699:10)'
2018-12-25 16:01:13,600 ERROR b'    at Module.load (internal/modules/cjs/loader.js:598:32)'
2018-12-25 16:01:13,600 ERROR b'    at tryModuleLoad (internal/modules/cjs/loader.js:537:12)'
2018-12-25 16:01:13,600 ERROR b'    at Function.Module._load (internal/modules/cjs/loader.js:529:3)'
2018-12-25 16:01:13,600 INFO Finished simserver stdout
2018-12-25 16:01:13,601 INFO Finished simserver stderr
2018-12-25 16:01:14,516 INFO Waiting for simserver stdout to finish
2018-12-25 16:01:14,517 INFO Waiting for simserver stderr to finish
```
I have followed the 'I get an error from the socketIO_client package' in https://github.com/minosworld/minos/blob/master/FAQ.md. But it didn't work.
Please give me some help!
Thank you very much!

angelxuanchang, 2018-12-25:
It looks that not all the node dependencies are installed correctly.  Please try to delete `node_modules` in the `server` directory and run `npm install` again and make sure you don't get any errors.  Please run `node --version` and let us know what node version you are using.

leah0206, 2018-12-26:
Hi, @angelxuanchang . Thanks for your help!
I have deleted `node_modules` in the `server` directory and run `npm install` again. But I encountered the following problem:
```
ccs@ccs-MS-7996:~/minos-new/minos/server$ npm install

> sstk@github:smartscenes/sstk#3735e535dd4e7d7b006c197e14e9bdef356f28c8 postinstall /home/ccs/minos-new/minos/server/node_modules/sstk
> node getdata.js && npm run bootstrap

正克隆到 'metadata'...
fatal: unable to access 'https://github.com/smartscenes/sstk-metadata.git/': Failed to receive SOCKS4 connect request ack.
ln: Source file does not exist
ln: Source file does not exist
ln: Source file does not exist
ln: Source file does not exist
ln: Source file does not exist
ln: Source file does not exist
ln: ENOENT: no such file or directory, symlink 'suncg.planner5d.models.v1.csv' -> '/home/ccs/minos-new/minos/server/node_modules/sstk/server/static/data/suncg/suncg.planner5d.models.full.csv'

> sstk@0.6.0 bootstrap /home/ccs/minos-new/minos/server/node_modules/sstk
> lerna bootstrap --hoist

lerna info version 2.11.0
lerna info Bootstrapping 3 packages
lerna info lifecycle preinstall
lerna info Installing external dependencies
lerna info hoist Installing hoisted dependencies into root
lerna info hoist Pruning hoisted dependencies
lerna info hoist Finished pruning hoisted dependencies
lerna ERR! execute callback with error
lerna ERR! Error: Command failed: npm install
lerna ERR! npm WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning ETIMEDOUT: request to https://registry.npmjs.org/body-parser failed, reason: connect ETIMEDOUT 104.16.21.35:443
lerna ERR! npm WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation.
lerna ERR! npm WARN deprecated filesaverjs@0.2.2: use 'file-saver' instead
lerna ERR! npm WARN deprecated jade@1.11.0: Jade has been renamed to pug, please install the latest version of pug instead of jade
lerna ERR! npm WARN deprecated constantinople@3.0.2: Please update to at least constantinople 3.1.1
lerna ERR! npm WARN deprecated transformers@2.1.0: Deprecated, use jstransformer
lerna ERR! npm WARN deprecated jscs-preset-wikimedia@1.0.1: Deprecated in favour of eslint-config-wikimedia. -- https://phabricator.wikimedia.org/T118941
lerna ERR! npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install the latest version of pug instead of jade
lerna ERR! npm WARN deprecated to-iso-string@0.0.2: to-iso-string has been deprecated, use @segment/to-iso-string instead.
lerna ERR! npm WARN deprecated nomnom@1.8.1: Package no longer supported. Contact support@npmjs.com for more info.
lerna ERR! npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
lerna ERR! npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
lerna ERR! npm WARN deprecated minimatch@2.0.10: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
lerna ERR! npm WARN deprecated hoek@0.7.6: The major version is no longer supported. Please update to 4.x or newer
lerna ERR! npm WARN deprecated coffee-script@1.12.7: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
lerna ERR! npm WARN deprecated hoek@5.0.4: This version is no longer maintained. Please upgrade to the latest version.
lerna ERR! npm ERR! Error while executing:
lerna ERR! npm ERR! /usr/bin/git ls-remote -h -t https://github.com/stanford-gfx/node-http-proxy.git
lerna ERR! npm ERR! 
lerna ERR! npm ERR! fatal: unable to access 'https://github.com/stanford-gfx/node-http-proxy.git/': Failed to receive SOCKS4 connect request ack.
lerna ERR! npm ERR! 
lerna ERR! npm ERR! exited with error code: 128
lerna ERR! 
lerna ERR! npm ERR! A complete log of this run can be found in:
lerna ERR! npm ERR!     /home/ccs/.npm/_logs/2018-12-26T03_00_44_868Z-debug.log
lerna ERR! 
lerna ERR!     at Promise.all.then.arr (/home/ccs/minos-new/minos/server/node_modules/execa/index.js:236:11)
{ Error: Command failed: npm install
npm WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning ETIMEDOUT: request to https://registry.npmjs.org/body-parser failed, reason: connect ETIMEDOUT 104.16.21.35:443
npm WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation.
npm WARN deprecated filesaverjs@0.2.2: use 'file-saver' instead
npm WARN deprecated jade@1.11.0: Jade has been renamed to pug, please install the latest version of pug instead of jade
npm WARN deprecated constantinople@3.0.2: Please update to at least constantinople 3.1.1
npm WARN deprecated transformers@2.1.0: Deprecated, use jstransformer
npm WARN deprecated jscs-preset-wikimedia@1.0.1: Deprecated in favour of eslint-config-wikimedia. -- https://phabricator.wikimedia.org/T118941
npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install the latest version of pug instead of jade
npm WARN deprecated to-iso-string@0.0.2: to-iso-string has been deprecated, use @segment/to-iso-string instead.
npm WARN deprecated nomnom@1.8.1: Package no longer supported. Contact support@npmjs.com for more info.
npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
[... 26 lines cut ...]
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! sstk@0.6.0 bootstrap: `lerna bootstrap --hoist`
npm ERR! Exit status 1
npm ERR! 
[... cut ...]
npm ERR! sstk@github:smartscenes/sstk#3735e535dd4e7d7b006c197e14e9bdef356f28c8 postinstall: `node getdata.js && npm run bootstrap`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the sstk@github:smartscenes/sstk#3735e535dd4e7d7b006c197e14e9bdef356f28c8 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ccs/.npm/_logs/2018-12-26T03_00_45_257Z-debug.log
```
I tried to run `NODE_TLS_REJECT_UNAUTHORIZED=0 npm install`. But it also didn't work!
Please help me!
Thank you very much!

angelxuanchang, 2018-12-27:
It looks like you are having trying using git clone to access some dependencies that are pulled down using https.  You can try to manually test `git clone https://github.com/smartscenes/sstk-metadata.git` to see if you get the same error (according to some websites, you can try doing `sudo` to see if that error goes away)

ngthanhtin, 2019-05-30:
did you solve the problem ?
I've encountered the same problem about SocketIO as you

morning197821, 2019-08-23:
Hello,@angelxuanchang  I wanted to try to run python3 -m minos.tools.pygame_client. However, I encountered the following problem:

> $ sudo python3 -m minos.tools.pygame_client --empty_room
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
fatal: Not a git repository (or any parent directory)：.git
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/project/minos/minos-master/minos/tools/pygame_client.py", line 457, in <module>
    main()
  File "/usr/project/minos/minos-master/minos/tools/pygame_client.py", line 415, in main
    sim = Simulator(vars(args))
  File "/usr/project/minos/minos-master/minos/lib/Simulator.py", line 110, in __init__
    cwd=stk_sim_path).rstrip(),
  File "/usr/lib/python3.6/subprocess.py", line 356, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['git', 'rev-parse', '--short', 'HEAD']' returned non-zero exit status 128.
2019-08-23 10:09:59,795 INFO sim00:Stopping the simulator
2019-08-23 10:09:59,796 INFO Counter()
2019-08-23 10:09:59,796 INFO sim00:Stopping child servers
2019-08-23 10:09:59,796 INFO sim00:Stopped child servers
2019-08-23 10:09:59,796 INFO sim00:Simulator killed.

 PLease give me some help. 
Thank you very much.