minosworld/minos #148: How to start the initial starting location to specific floor of matter port house ID.

DRAhmadFaraz, 2019-01-16:
If any matterport house has more then 1 floor then How to start the initial starting location to specific floor of matter port house ID.? as shown as house ID "1LXtFkjw3qL"  has 3 floors and I want my initial position to specific floor (ground floor) rather then random floor choose.

[image]
[image]
[image]

DRAhmadFaraz, 2019-01-16:
@angelxuanchang

angelxuanchang, 2019-01-17:
Please try the latest v0.7.x branch to specify the level (starting at 0) when running `pygame_client` as follows:

`python3 -m minos.tools.pygame_client  --depth  -s map --navmap --scene_ids 1LXtFkjw3qL --level 2 --dataset mp3d --env_config pointgoal_mp3d_s --log_action_trace true`

DRAhmadFaraz, 2019-01-17:
I am using 0.7.x version and get this error 

pygame_client.py: error: unrecognized arguments: --level 2
or
pygame_client.py: error: unrecognized arguments: --level 0

[image]

angelxuanchang, 2019-01-17:
Please make sure you have checked out the latest 0.7x branch.