minosworld/minos #150: Errors with Socket IO issue and missing 'suncg.materials.csv'

jsdd25, 2019-01-28:
Hi,

I have downloaded SUNCG in `$HOME/work/suncg` and tried to run MINOS with `python3 main.py --env_type indoor --env_name pointgoal_suncg_se`. After I run that command, the following errors occur consecutively:
`2019-01-28 20:27:40,149 WARNING localhost:51482/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=51482): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1548678460146-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fd54626b7f0>: Failed to establish a new connection: [Errno 111] Connection refused',))`

I am using 0.7.x branch. My platform is Linux - Ubuntu 16.04 LTS with Python 3.6.4 |Anaconda custom (64-bit). The socketio version is `socketIO-client-2         0.7.4` and `websocket-client          0.54.0`. 
Node version: 'v10.13.0'. Npm version: '6.4.1'.

here is my logs/sim01/ simserver.logs file status. kindly guide me what to do...
`2019-01-28 20:12:26,093 INFO b'Configuration { fileCache: { size: 50 },'
2019-01-28 20:12:26,094 INFO b'  imageCache: { size: 200 },'
2019-01-28 20:12:26,094 INFO b'  imageQueue: { concurrency: 4 },'
2019-01-28 20:12:26,094 INFO b"  base_url: '/home/jin/work/',"
2019-01-28 20:12:26,094 INFO b'  assets_url:'
2019-01-28 20:12:26,094 INFO b"   'file:///home/jin/minos/minos/server/node_modules/sstk/ssc/../server/static' }"
2019-01-28 20:12:26,252 INFO b'Waiting for client connection on port 32286'
2019-01-28 20:12:26,264 ERROR b'Cannot register unknown asset 2d3ds'
2019-01-28 20:12:26,264 INFO b'register ../server/static/data/matterport/matterport3d.json'
2019-01-28 20:12:26,266 INFO b'Registered asset group: mp3d'
2019-01-28 20:12:26,268 INFO b'Got 90 assets'
2019-01-28 20:12:26,308 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2019-01-28 20:12:26,309 INFO b'Replace asset group p5d'
2019-01-28 20:12:26,309 INFO b'Registered asset group: p5d'
2019-01-28 20:12:26,361 INFO b'Got 2623 assets'
2019-01-28 20:12:26,933 INFO b'Loaded lightSpecs for 202 models.'
2019-01-28 20:12:26,933 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2019-01-28 20:12:26,935 INFO b'Replace asset group p5dTexture'
2019-01-28 20:12:26,935 INFO b'Registered asset group: p5dTexture'
2019-01-28 20:12:27,142 INFO b'Got 67072 assets'
2019-01-28 20:12:27,257 INFO b'Client wv_dFQLBvSWLuSs3AAAA connected on port 32286'
2019-01-28 20:12:27,338 INFO b'THREE.WebGLRenderer 84'
2019-01-28 20:12:27,339 ERROR b'THREE.WebGLRenderer: WEBGL_depth_texture extension not supported.'
2019-01-28 20:12:27,339 ERROR b'THREE.WebGLRenderer: OES_texture_float_linear extension not supported.'
2019-01-28 20:12:27,339 ERROR b'THREE.WebGLRenderer: OES_texture_half_float extension not supported.'
2019-01-28 20:12:27,339 ERROR b'THREE.WebGLRenderer: OES_texture_half_float_linear extension not supported.'
2019-01-28 20:12:27,339 ERROR b'THREE.WebGLRenderer: OES_standard_derivatives extension not supported.'
2019-01-28 20:12:27,367 ERROR b'[sensors] colorEncoding rgba is set, overriding encoding gray'
2019-01-28 20:12:27,375 INFO b"Creating Simulator with options { simulator: 'room_simulator',"
2019-01-28 20:12:27,375 INFO b'  numSimulators: 1,'
2019-01-28 20:12:27,375 INFO b"  modalities: [ 'color', 'measurements' ],"
2019-01-28 20:12:27,375 INFO b"  outputs: [ 'color', 'measurements', 'rewards', 'terminals' ],"
2019-01-28 20:12:27,375 INFO b'  resolution: [ 84, 84 ],'
2019-01-28 20:12:27,375 INFO b'  frameSkip: 1,'
2019-01-28 20:12:27,375 INFO b"  host: 'localhost',"
2019-01-28 20:12:27,375 INFO b'  logActionTrace: false,'
2019-01-28 20:12:27,375 INFO b'  autoStart: true,'
2019-01-28 20:12:27,375 INFO b"  collisionDetection: { mode: 'navgrid' },"
2019-01-28 20:12:27,375 INFO b'  navmap:'
2019-01-28 20:12:27,375 INFO b'   { refineGrid: true,'
2019-01-28 20:12:27,375 INFO b'     autoUpdate: true,'
2019-01-28 20:12:27,375 INFO b'     allowDiagonalMoves: true,'
2019-01-28 20:12:27,375 INFO b'     reverseEdgeOrder: false },'
2019-01-28 20:12:27,376 INFO b"  rewardType: 'dist_time',"
2019-01-28 20:12:27,376 INFO b'  observations:'
2019-01-28 20:12:27,376 INFO b'   { color: true,'
2019-01-28 20:12:27,376 INFO b'     forces: false,'
2019-01-28 20:12:27,376 INFO b'     audio: false,'
2019-01-28 20:12:27,376 INFO b'     normal: false,'
2019-01-28 20:12:27,376 INFO b'     depth: false,'
2019-01-28 20:12:27,376 INFO b'     objectId: false,'
2019-01-28 20:12:27,376 INFO b'     objectType: false,'
2019-01-28 20:12:27,376 INFO b'     roomType: false,'
2019-01-28 20:12:27,376 INFO b'     roomId: false,'
2019-01-28 20:12:27,376 INFO b'     map: false,'
2019-01-28 20:12:27,376 INFO b'     rgb: true },'
2019-01-28 20:12:27,376 INFO b"  colorEncoding: 'rgba',"
2019-01-28 20:12:27,376 INFO b'  scene:'
2019-01-28 20:12:27,376 INFO b'   { archOnly: false,'
2019-01-28 20:12:27,376 INFO b'     retexture: true,'
2019-01-28 20:12:27,376 INFO b'     emptyRoom: true,'
[... 729 lines cut ...]
  },
  "dependencies": {
    "lerna": "^2.11.0",
    "shelljs": "^0.7.0"
  },
  "description": "SmartScenes ToolKit: a library for viewing, editing and annotating 3D models and scenes",
  "homepage": "https://github.com/smartscenes/sstk",
  "license": "MIT",
  "main": "stk-ssc",
  "name": "sstk",
  "repository": {
    "type": "git",
    "url": "git+https://github.com/smartscenes/sstk.git"
  },
  "scripts": {
    "bootstrap": "lerna bootstrap --hoist",
    "clean": "lerna clean",
    "ls": "lerna ls",
    "postinstall": "node getdata.js && npm run bootstrap",
    "test": "lerna run test"
  },
  "version": "0.6.0"
}`
Please help me out here.
Many thanks once again!

angelxuanchang, 2019-01-30:
Thanks for reporting this issue.  We have made a small fix.  Please try again by refreshing the sstk modules in the minos/server directory

```
cd minos/server
rm package-lock.json
rm -rf node_modules/sstk
npm install
```

jsdd25, 2019-02-01:
Thanks @angelxuanchang .
I have followed your instructions. However, it didn't work.
here is my logs/sim01/ simserver.logs file status. kindly guide me what to do...
```
2019-02-01 18:43:50,968 INFO b'Configuration { fileCache: { size: 50 },'
2019-02-01 18:43:50,968 INFO b'  imageCache: { size: 200 },'
2019-02-01 18:43:50,968 INFO b'  imageQueue: { concurrency: 4 },'
2019-02-01 18:43:50,968 INFO b"  base_url: '/home/jin/work/',"
2019-02-01 18:43:50,968 INFO b'  assets_url:'
2019-02-01 18:43:50,968 INFO b"   'file:///home/jin/minos/minos/server/node_modules/sstk/ssc/../server/static' }"
2019-02-01 18:43:51,123 INFO b'Waiting for client connection on port 45186'
2019-02-01 18:43:51,134 ERROR b'Cannot register unknown asset 2d3ds'
2019-02-01 18:43:51,135 INFO b'register ../server/static/data/matterport/matterport3d.json'
2019-02-01 18:43:51,136 INFO b'Registered asset group: mp3d'
2019-02-01 18:43:51,139 INFO b'Got 90 assets'
2019-02-01 18:43:51,182 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2019-02-01 18:43:51,182 INFO b'Replace asset group p5d'
2019-02-01 18:43:51,182 INFO b'Registered asset group: p5d'
2019-02-01 18:43:51,235 INFO b'Got 2623 assets'
2019-02-01 18:43:51,800 INFO b'Loaded lightSpecs for 202 models.'
2019-02-01 18:43:51,801 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2019-02-01 18:43:51,803 INFO b'Replace asset group p5dTexture'
2019-02-01 18:43:51,803 INFO b'Registered asset group: p5dTexture'
2019-02-01 18:43:52,007 INFO b'Got 67072 assets'
2019-02-01 18:43:52,140 INFO b'Client 5JDPFwye7JyizwxSAAAA connected on port 45186'
2019-02-01 18:43:52,142 INFO b'Loaded LightSpecs for 202 models'
2019-02-01 18:43:52,261 INFO b'THREE.WebGLRenderer 84'
2019-02-01 18:43:52,261 ERROR b'THREE.WebGLRenderer: WEBGL_depth_texture extension not supported.'
2019-02-01 18:43:52,261 ERROR b'THREE.WebGLRenderer: OES_texture_float_linear extension not supported.'
2019-02-01 18:43:52,261 ERROR b'THREE.WebGLRenderer: OES_texture_half_float extension not supported.'
2019-02-01 18:43:52,262 ERROR b'THREE.WebGLRenderer: OES_texture_half_float_linear extension not supported.'
2019-02-01 18:43:52,262 ERROR b'THREE.WebGLRenderer: OES_standard_derivatives extension not supported.'
2019-02-01 18:43:52,268 INFO b"Creating Simulator with options { simulator: 'room_simulator',"
2019-02-01 18:43:52,268 INFO b'  numSimulators: 1,'
2019-02-01 18:43:52,268 INFO b"  modalities: [ 'color', 'measurements' ],"
2019-02-01 18:43:52,268 INFO b"  outputs: [ 'color', 'measurements', 'rewards', 'terminals' ],"
2019-02-01 18:43:52,268 INFO b'  resolution: [ 84, 84 ],'
2019-02-01 18:43:52,268 INFO b'  frameSkip: 1,'
2019-02-01 18:43:52,269 INFO b"  host: 'localhost',"
2019-02-01 18:43:52,269 INFO b'  logActionTrace: false,'
2019-02-01 18:43:52,269 INFO b'  autoStart: true,'
2019-02-01 18:43:52,269 INFO b"  collisionDetection: { mode: 'navgrid' },"
2019-02-01 18:43:52,269 INFO b'  navmap:'
2019-02-01 18:43:52,269 INFO b'   { refineGrid: true,'
2019-02-01 18:43:52,269 INFO b'     autoUpdate: true,'
2019-02-01 18:43:52,269 INFO b'     allowDiagonalMoves: true,'
2019-02-01 18:43:52,269 INFO b'     reverseEdgeOrder: false },'
2019-02-01 18:43:52,270 INFO b"  rewardType: 'dist_time',"
2019-02-01 18:43:52,270 INFO b'  observations:'
2019-02-01 18:43:52,270 INFO b'   { color: true,'
2019-02-01 18:43:52,270 INFO b'     forces: false,'
2019-02-01 18:43:52,270 INFO b'     audio: false,'
2019-02-01 18:43:52,270 INFO b'     normal: false,'
2019-02-01 18:43:52,270 INFO b'     depth: false,'
2019-02-01 18:43:52,270 ERROR b'[sensors] colorEncoding rgba is set, overriding encoding gray'
2019-02-01 18:43:52,271 INFO b'     objectId: false,'
2019-02-01 18:43:52,271 INFO b'     objectType: false,'
2019-02-01 18:43:52,271 INFO b'     roomType: false,'
2019-02-01 18:43:52,271 INFO b'     roomId: false,'
2019-02-01 18:43:52,271 INFO b'     map: false,'
2019-02-01 18:43:52,271 INFO b'     rgb: true },'
2019-02-01 18:43:52,271 INFO b"  colorEncoding: 'rgba',"
2019-02-01 18:43:52,271 INFO b'  scene:'
2019-02-01 18:43:52,271 INFO b'   { archOnly: false,'
2019-02-01 18:43:52,271 INFO b'     retexture: true,'
2019-02-01 18:43:52,271 INFO b'     emptyRoom: true,'
2019-02-01 18:43:52,271 INFO b"     dataset: 'p5dScene',"
2019-02-01 18:43:52,271 INFO b'     replaceModels:'
2019-02-01 18:43:52,271 INFO b"      { '73': '73_2',"
2019-02-01 18:43:52,271 INFO b"        '122': '122_2',"
[... 287 lines cut ...]
2019-02-01 18:43:54,199 INFO b'imported material indices'
2019-02-01 18:43:56,311 INFO b'imported material counts'
2019-02-01 18:43:56,312 INFO b'Removing subscribers for event ready [ 0 ]'
2019-02-01 18:43:56,312 INFO b'importMaterialCsvs: 3965.560ms'
2019-02-01 18:43:56,314 INFO b'Loading scene  p5dScene.40666d3bde3d0c0dba48550a5e24314a'
2019-02-01 18:43:56,328 ERROR b'/home/jin/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:108260'
2019-02-01 18:43:56,328 ERROR b'\t  if (this.opts.scene && this.opts.scene.fullId.startsWith("mp3d")) {'
2019-02-01 18:43:56,328 ERROR b'\t                                                ^'
2019-02-01 18:43:56,328 ERROR b''
2019-02-01 18:43:56,328 ERROR b"TypeError: Cannot read property 'startsWith' of undefined"
2019-02-01 18:43:56,328 ERROR b'    at Simulator.__loadScene (/home/jin/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:108260:50)'
[... cut ...]
2019-02-01 18:43:56,328 ERROR b'    at Timeout._onTimeout (/home/jin/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:108129:35)'
2019-02-01 18:43:56,328 ERROR b'    at ontimeout (timers.js:436:11)'
2019-02-01 18:43:56,328 ERROR b'    at tryOnTimeout (timers.js:300:5)'
2019-02-01 18:43:56,328 ERROR b'    at listOnTimeout (timers.js:263:5)'
2019-02-01 18:43:56,380 INFO Finished simserver stderr
2019-02-01 18:43:56,380 INFO Finished simserver stdout
```
Please help me out here.
Thank you very much!

angelxuanchang, 2019-02-01:
Sorry, I was confused which version of minos you were using.  This issue is also fixed for the v0.6.x branch.  Please checkout the latest v0.6.x (or master) branch of minos, make sure to do `npm install` in `minos/server` and try again and let us know if you still have issues.

jsdd25, 2019-02-02:
I have fixed this problem. Thank you very much!!!

closed 2019-02-02 by jsdd25

chenwydj, 2019-07-19:
@jsdd25 How did you solve this problem? I also met the error 'THREE.WebGLRenderer: Error creating WebGL context.'
Thanks!!!