minosworld/minos #153: Error on SocketIO when running `python3 -m minos.tools.pygame_client --empty_room`

chrischoy, 2019-02-12:
Hi Angel,

I was trying to use minos, but I got this error when I run ` python3 -m minos.tools.pygame_client --empty_room`.
It was compiled on the latest master.

```
(py3-minos-master) ➜ choychri@choychri-desk1 ~/projects/minos_world/minos git:(master) ✗ python3 -m minos.tools.pygame_client --empty_room   {19-02-12 15:41}
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
2019-02-12 15:41:27,520 INFO {'sim_id': 'sim00', 'machine': 'choychri-desk1', 'sim_git_hash': 'ce7b7c2', 'stk_git_hash': 'ce7b7c2'}
Starting simulator...
2019-02-12 15:41:27,535 INFO sim00:Starting sim server at /home/choychri/projects/minos_world/minos/minos/server/server.js with port 25826
2019-02-12 15:41:29,589 INFO sim00:connect
Traceback (most recent call last):
  File "/home/choychri/projects/minos_world/minos/minos/tools/pygame_client.py", line 438, in main
    ep_info = sim.start()
  File "/home/choychri/projects/minos_world/minos/minos/lib/Simulator.py", line 330, in start
    self._rpc('start', self.params, self.on_started)
  File "/home/choychri/projects/minos_world/minos/minos/lib/Simulator.py", line 186, in _rpc
    return rpc.call(name, data, callback, seconds, check_wait=lambda: self.running)
  File "/home/choychri/projects/minos_world/minos/minos/lib/util/RpcCall.py", line 25, in call
    self.sio.wait_for_callbacks(seconds=seconds)    # wait for response
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/__init__.py", line 450, in wait_for_callbacks
    self.wait(seconds, for_callbacks=True)
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/__init__.py", line 245, in wait
    self._process_packets()
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/__init__.py", line 269, in _process_packets
    self._process_packet(engineIO_packet)
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/__init__.py", line 471, in _process_packet
    self.placeholder.add(engineIO_packet_data)
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/parsers.py", line 35, in add
    self.replace_placeholders()
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/parsers.py", line 44, in replace_placeholders
    self.args = traverse(deepcopy(self.args), predicate, fn)
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/parsers.py", line 57, in traverse
    obj[i] = traverse(value, predicate, fn)
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/parsers.py", line 52, in traverse
    obj[key] = traverse(value, predicate, fn)
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/parsers.py", line 52, in traverse
    obj[key] = traverse(value, predicate, fn)
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/parsers.py", line 52, in traverse
    obj[key] = traverse(value, predicate, fn)
  [Previous line repeated 1 more time]
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/parsers.py", line 57, in traverse
    obj[i] = traverse(value, predicate, fn)
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/parsers.py", line 48, in traverse
    if predicate(obj):
  File "/home/choychri/anaconda3/envs/py3-minos-master/lib/python3.7/site-packages/socketIO_client/parsers.py", line 39, in predicate
    return '_placeholder' in obj and 'num' in obj
TypeError: argument of type 'float' is not iterable
Error running simulator. Aborting.
2019-02-12 15:41:30,562 INFO sim00:Stopping the simulator
2019-02-12 15:41:30,563 INFO Counter()
```

chrischoy, 2019-02-12:
Directly cloning the SocketIO from the main repository seems to solve the problem.

```
pip uninstall socketIO-client-2 -y
git clone https://github.com/msavva/socketIO-client-2
cd socketIO-client-2
pip install -e .
```

jcboxq, 2021-04-06:
> Directly cloning the SocketIO from the main repository seems to solve the problem.
> 
> ```
> pip uninstall socketIO-client-2 -y
> git clone https://github.com/msavva/socketIO-client-2
> cd socketIO-client-2
> pip install -e .
> ```
> 
> If you have the socketIO-client-2 repository already, before you `pip install -e .` remove `socketIO_client_2.egg-info` first.

Yes, it works! This just saved me out.