minosworld/minos #157: From wher  'agent_state': in code is being generated ?

Usama323, 2019-04-04:
Can you kindly point out portion of code responsible to generate ''agent_state': when I press 'O'. Thanks in advance for this act of kindness

Usama323, 2019-04-08:
Someone kindly reply to this question, I need its answer fast

msavva, 2019-04-09:
Hi @Usama323 , apologies for the delay.  You can find the relevant code in https://github.com/smartscenes/sstk/blob/master/client/js/lib/sim/Agent.js#L170-L175

Usama323, 2019-04-09:
Thanks @msavva for fast reply,  can you kindly point out portion in this code where values of " x, y, z, ox, oy & oz" are being generated? I want to display them with out pressing 'O' ( continuous displaying if action is changed)

Usama323, 2019-04-12:
Kindly reply to the last comment !!

Jemmagu, 2019-04-14:
excuse me, do you know how can I get access to the SUNCG dataset? it seems unavailable on the Princeton website and the agreement form has been cancelled. Please Kindly tell me how can I get access to the dataset !!

DRAhmadFaraz, 2019-04-15:
I am also facing these problems so please I need its answer as soon as possible. 
Regards