minosworld/minos #158: Portion of code generating 'action_trace.csv' file ?

Usama323, 2019-04-04:
I want to know the portion of code that is responsible to generate/save "action_trace" file. Thanks in advance for this act of kindness

Usama323, 2019-04-08:
Someone kindly reply to this question, I need its answer fast

msavva, 2019-04-09:
Hi @Usama323 , apologies for the delay.  The relevant code is at https://github.com/smartscenes/sstk/blob/master/client/js/lib/sim/ActionTraceLog.js

Usama323, 2019-04-09:
Thanks @msavva for fast reply,  can you kindly point out portion in this code where values of " x, y , z, ox, oy and oz" are being written to ".csv" file ! I want to see the portion where you are combing three angles and saving it as one angle. Thanks in advance for this act of kindness

Usama323, 2019-04-12:
Kindly reply to the last comment !!

DRAhmadFaraz, 2019-04-15:
I am also facing these problems so please I need its answer as soon as possible. 
Regards