minosworld/minos #159: How I can download Suncg data set, it's unavailable on the website!

Jemmagu, 2019-04-14:
How I can download Suncg data set, it seems unavailable on the Princeton website  and the agreement google docs has been cancelled. Please Kindly tell me how can I get access to the dataset. quite urgent!!

Usama323, 2019-04-14:
Here is the issue kindly see this!

https://github.com/minosworld/minos/issues/94

hzxie, 2019-04-25:
@Usama323 
However, the links are unavailable now. 
I sent several emails to the team, but they didn't reply.

Usama323, 2019-04-27:
@hzxie  I have rechecked it, it is gone now. I also need that data, any author kindly respond to this problem as soon as possible! @msavva  @angelxuanchang

tonyromarock, 2019-06-12:
I do not think that the data set will be made available again anytime soon: https://www.law.com/therecorder/2019/06/06/home-design-site-claims-facebook-princeton-nabbed-data-for-3-d-objects/

gyzhang123, 2019-07-22:
can we get some samples like that ?

closed 2019-08-03 by Jemmagu