minosworld/minos #165: What is the format of grid.json file in MATTERPORT 3D ?

ngthanhtin, 2019-06-24:
Hi, I am writing a Path Finding algorithm for minos but I've encountered two issues:
1) I don't understand the format of grid.json file in MATTERPORT 3d, for example: 17DRP5sb8fy.furnished.json. I want to know this because I want to get the map information (where is occupied, where is not) to pass to the Path Finding algorithm.
2) I want to know how to convert integer positions I've found from Path Finding algorithm to minos coordinate (it means the floating-point coordinates).

Thanks in advance.

ngthanhtin, 2019-06-25:
@msavva I see in the https://github.com/minosworld/minos/wiki/NavGrid-format link, but I don't know occupancy tag means, is it the map ? If it is, what the format of it because I think the map is 3D but the data of occupancy is just integers.

angelxuanchang, 2019-07-02:
The occupancy is a binary grid with 0 indicating that it is free-space and 1 that it is occupied.  Please note that it uses a simple run-length encoding (array of pairs of value followed by number of times the value appears).

ngthanhtin, 2019-07-03:
thank you very much

closed 2019-07-05 by ngthanhtin