minosworld/minos #167: Wrong shortest path

shsjxzh, 2019-08-01:
Hi, I find that some shortest path shown in the nav map requires me to go through the closed door, which cannot be done temporarily. How can I get a correct shortest path?

You can run " python3 -m minos.tools.pygame_client --dataset mp3d --scene_ids 17DRP5sb8fy --env_config pointgoal_mp3d_s -s map --navmap" and sometimes it will show wrong path.

[image]

angelxuanchang, 2019-08-09:
Thanks for reporting this.  This is an error with having some configuration overridden by the --navmap option.  Please try the updated 0.7.x branch with:
`python3 -m minos.tools.pygame_client --dataset mp3d --scene_ids 17DRP5sb8fy --env_config pointgoal_mp3d_s -s map --navmap--collision_mode navgrid`

shsjxzh, 2019-11-17:
@angelxuanchang Sorry for late response! The 0.7.x branch really works. However, when I try to run "gym/demo.py", I find that there is something wrong with this recent modification. I modify the code in the following way:

  if 'navmap' in dir(args):
        if args.navmap and type(args.navmap) == bool and sim_config.sim_defaults.get('navmap'):
            args.navmap = None