minosworld/minos #169: connection refused

UncleMEDM, 2019-08-14:
When I run the following command "python3 -m minos.tools.pygame_client --dataset mp3d --scene_ids 17DRP5sb8fy --env_config pointgoal_mp3d_s". The following error appeared:

(minos) haolin@local:/data/haolin/minos$ python3 -m minos.tools.pygame_client --dataset mp3d --scene_ids 17DRP5sb8fy --env_config pointgoal_mp3d_s
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
2019-08-15 00:17:22,030 INFO {'sim_id': 'sim00', 'machine': 'local', 'sim_git_hash': 'ce7b7c2', 'stk_git_hash': 'ce7b7c2'}
Starting simulator...
2019-08-15 00:17:22,050 INFO sim00:Starting sim server at /data/haolin/minos/minos/server/server.js with port 6002
2019-08-15 00:17:23,063 WARNING localhost:6002/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=6002): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1565799443058-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f34b1a32c50>: Failed to establish a new connection: [Errno 111] Connection refused',))

I don't know how to solve this. Below is the log file for the server:

2019-08-15 00:13:33,012 INFO b'Configuration { fileCache: { size: 50 },'
2019-08-15 00:13:33,012 INFO b'  imageCache: { size: 200 },'
2019-08-15 00:13:33,012 INFO b'  imageQueue: { concurrency: 4 },'
2019-08-15 00:13:33,012 INFO b"  base_url: '/home/haolin/work/',"
2019-08-15 00:13:33,012 INFO b'  assets_url:'
2019-08-15 00:13:33,012 INFO b"   'file:///data/haolin/minos/minos/server/node_modules/sstk/ssc/../server/static' }"
2019-08-15 00:13:33,339 INFO b'Waiting for client connection on port undefined'
2019-08-15 00:13:33,366 ERROR b'Cannot register unknown asset 2d3ds'
2019-08-15 00:13:33,367 INFO b'register ../server/static/data/matterport/matterport3d.json'
2019-08-15 00:13:33,370 INFO b'Registered asset group: mp3d'
2019-08-15 00:13:33,374 INFO b'Got 90 assets'
2019-08-15 00:13:33,456 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2019-08-15 00:13:33,457 INFO b'Replace asset group p5d'
2019-08-15 00:13:33,457 INFO b'Registered asset group: p5d'
2019-08-15 00:13:33,553 INFO b'Got 2623 assets'
2019-08-15 00:13:34,699 INFO b'Loaded lightSpecs for 202 models.'
2019-08-15 00:13:34,699 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2019-08-15 00:13:34,704 INFO b'Replace asset group p5dTexture'
2019-08-15 00:13:34,704 INFO b'Registered asset group: p5dTexture'
2019-08-15 00:13:35,084 INFO b'Got 67072 assets'
2019-08-15 00:13:35,299 INFO b'Loaded LightSpecs for 202 models'
2019-08-15 00:13:35,315 INFO Finished simserver stderr
2019-08-15 00:13:35,315 INFO Finished simserver stdout
2019-08-15 00:14:13,975 INFO Waiting for simserver stdout to finish
2019-08-15 00:14:13,975 INFO Waiting for simserver stderr to finish

and below is the log file for the simulator:

2019-08-15 00:13:31,581 INFO {'sim_id': 'sim00', 'machine': 'local', 'sim_git_hash': 'ce7b7c2', 'stk_git_hash': 'ce7b7c2'}
2019-08-15 00:13:31,602 INFO sim00:Starting sim server at /data/haolin/minos/minos/server/server.js with port 6002
2019-08-15 00:14:13,974 INFO sim00:Stopping the simulator
2019-08-15 00:14:13,974 INFO Counter()
2019-08-15 00:14:13,975 INFO sim00:Stopping child servers
2019-08-15 00:14:13,975 INFO sim00:Killing sim pid 2916
2019-08-15 00:14:13,975 INFO sim00:Stopped child servers
2019-08-15 00:14:13,975 INFO sim00:Simulator killed.

It seems like that everything was fine but the simulator cannot connect to the server.

angelxuanchang, 2019-08-14:
Please try the v0.7.x branch and let us know if you still get the same issue.

UncleMEDM, 2019-08-19:
Thank you for your help. The v0.7.x branch works for me, and the interactive viewer was successful. However, there are still several error message in the log file, I don't know if it would affect the program further, but it seems good so far.

The log message of the server is shown below, with several error messages associated with the WebGL: 
2019-08-19 10:06:52,132 INFO b'Configuration { fileCache: { size: 50 },'
2019-08-19 10:06:52,133 INFO b'  imageCache: { size: 200 },'
2019-08-19 10:06:52,133 INFO b'  imageQueue: { concurrency: 4 },'
2019-08-19 10:06:52,133 INFO b"  base_url: '/home/haolin/work/',"
2019-08-19 10:06:52,133 INFO b'  assets_url:'
2019-08-19 10:06:52,133 INFO b"   'file:///data/haolin/minos-0.7.x/minos/server/node_modules/sstk/ssc/../server/static' }"
2019-08-19 10:06:52,520 INFO b'Waiting for client connection on port 33991'
2019-08-19 10:06:52,548 INFO b'register ../server/static/data/matterport/matterport3d.json'
2019-08-19 10:06:52,551 INFO b'Registered asset group: mp3d'
2019-08-19 10:06:52,556 INFO b'Got 90 assets'
2019-08-19 10:06:52,653 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2019-08-19 10:06:52,655 INFO b'Replace asset group p5d'
2019-08-19 10:06:52,655 INFO b'Registered asset group: p5d'
2019-08-19 10:06:52,753 INFO b'Got 2623 assets'
2019-08-19 10:06:54,208 INFO b'Loaded lightSpecs for 202 models.'
2019-08-19 10:06:54,208 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2019-08-19 10:06:54,214 INFO b'Replace asset group p5dTexture'
2019-08-19 10:06:54,214 INFO b'Registered asset group: p5dTexture'
2019-08-19 10:06:54,584 INFO b'Got 67072 assets'
2019-08-19 10:06:54,797 INFO b'Client 6SOPAPnUWCoNpkCDAAAA connected on port 33991'
2019-08-19 10:06:54,809 INFO b'Loaded LightSpecs for 202 models'
2019-08-19 10:06:56,015 INFO b'THREE.WebGLRenderer 95'
2019-08-19 10:06:56,017 ERROR b'THREE.WebGLRenderer: WEBGL_depth_texture extension not supported.'
2019-08-19 10:06:56,017 ERROR b'THREE.WebGLRenderer: OES_texture_half_float extension not supported.'
2019-08-19 10:06:56,018 ERROR b'THREE.WebGLRenderer: OES_texture_half_float_linear extension not supported.'
2019-08-19 10:06:56,018 ERROR b'THREE.WebGLRenderer: OES_standard_derivatives extension not supported.'
2019-08-19 10:06:56,018 ERROR b'THREE.WebGLRenderer: OES_texture_float_linear extension not supported.'
2019-08-19 10:06:56,032 INFO b"Creating Simulator with options { simulator: 'room_simulator',"
2019-08-19 10:06:56,032 INFO b'  numSimulators: 1,'
2019-08-19 10:06:56,032 INFO b"  modalities: [ 'color', 'measurements' ],"
2019-08-19 10:06:56,033 INFO b"  outputs: [ 'color', 'measurements', 'rewards', 'terminals' ],"
2019-08-19 10:06:56,033 INFO b'  resolution: [ 84, 84 ],'
2019-08-19 10:06:56,033 INFO b'  frameSkip: 1,'
2019-08-19 10:06:56,033 INFO b"  host: 'localhost',"
2019-08-19 10:06:56,033 INFO b'  logActionTrace: false,'
2019-08-19 10:06:56,033 INFO b'  autoStart: true,'
2019-08-19 10:06:56,033 INFO b"  collisionDetection: { mode: 'raycast' },"
2019-08-19 10:06:56,033 INFO b'  navmap: false,'
2019-08-19 10:06:56,033 INFO b"  rewardType: 'dist_time',"
2019-08-19 10:06:56,033 INFO b'  observations:'
2019-08-19 10:06:56,033 INFO b'   { color: true,'
2019-08-19 10:06:56,033 INFO b'     depth: null,'
2019-08-19 10:06:56,033 INFO b'     forces: null,'
2019-08-19 10:06:56,033 INFO b'     audio: null,'
2019-08-19 10:06:56,033 INFO b'     rgb: true,'
2019-08-19 10:06:56,033 INFO b'     normal: false,'
2019-08-19 10:06:56,033 INFO b'     objectType: false,'
2019-08-19 10:06:56,033 INFO b'     objectId: false,'
2019-08-19 10:06:56,033 INFO b'     roomType: false,'
2019-08-19 10:06:56,033 INFO b'     roomId: false,'
2019-08-19 10:06:56,033 INFO b'     map: false },'
2019-08-19 10:06:56,033 INFO b"  colorEncoding: 'rgba',"
2019-08-19 10:06:56,033 INFO b'  scene:'
2019-08-19 10:06:56,034 INFO b"   { dataset: 'mp3d',"
2019-08-19 10:06:56,034 INFO b'     replaceModels:'
2019-08-19 10:06:56,034 INFO b"      { '73': '73_2',"
2019-08-19 10:06:56,034 INFO b"        '122': '122_2',"
2019-08-19 10:06:56,034 INFO b"        '133': '133_2',"
2019-08-19 10:06:56,034 INFO b"        '214': '214_2',"
2019-08-19 10:06:56,034 INFO b"        '246': '246_2',"
2019-08-19 10:06:56,034 INFO b"        '247': '247_2',"
2019-08-19 10:06:56,034 INFO b"        '326': '327',"
2019-08-19 10:06:56,034 INFO b"        '331': '327',"
2019-08-19 10:06:56,034 INFO b"        '502': '502_2',"
2019-08-19 10:06:56,034 INFO b"        '622': '625',"
2019-08-19 10:06:56,034 INFO b"        '756': '756_2',"
2019-08-19 10:06:56,034 INFO b"        '757': '757_2',"
[... 225 lines cut ...]
2019-08-19 10:06:56,111 INFO b'preloading /home/haolin/work//mp3d/scans/17DRP5sb8fy/house_segmentations/17DRP5sb8fy.house using HouseLoader'
2019-08-19 10:06:56,111 INFO b'preloading /home/haolin/work//mp3d/scans/17DRP5sb8fy/instances_mesh/bed1a77d92d64f5cbbaaae4feed64ec1.mtl using MTLLoaderWrapper'
2019-08-19 10:06:56,114 ERROR b"Error preloading objectInstanceMaterials for asset mp3d.17DRP5sb8fy /home/haolin/work//mp3d/scans/17DRP5sb8fy/instances_mesh/bed1a77d92d64f5cbbaaae4feed64ec1.mtl { [Error: ENOENT: no such file or directory, open '/home/haolin/work//mp3d/scans/17DRP5sb8fy/instances_mesh/bed1a77d92d64f5cbbaaae4feed64ec1.mtl'"
2019-08-19 10:06:56,114 ERROR b''
2019-08-19 10:06:56,114 ERROR b']'
2019-08-19 10:06:56,114 ERROR b'  errno: -2,'
2019-08-19 10:06:56,115 ERROR b"  code: 'ENOENT',"
[... cut ...]
2019-08-19 10:07:10,668 INFO b'Removing subscribers for event drain [ 0 ]'
2019-08-19 10:07:10,671 INFO b'Timing waitImages: 262.234ms'
2019-08-19 10:07:10,843 ERROR b'THREE.WebGLRenderer: EXT_texture_filter_anisotropic extension not supported.'
2019-08-19 10:07:11,111 INFO b'Timing start: 15016.014ms'
2019-08-19 10:07:56,375 INFO Waiting for simserver stdout to finish
2019-08-19 10:07:56,466 INFO Finished simserver stdout
2019-08-19 10:07:56,466 INFO Finished simserver stderr
2019-08-19 10:07:56,467 INFO Waiting for simserver stderr to finish